ARMmbed/mbed-os #13024: crash on STM32F429 with modem gemalto

dqdqdq31, 2020-05-28:
Run mbed-cloud-client-example project with mbed-os-6.0.0-beta-1 and it crashed.
But it works with mbed-os-6.0.0-alpha-3. (all the steps and settings are the same)

Target:
STM32F429 with Gemalto ELS-61

Here are the steps:
1. import mbed-cloud-client-example project from "https://github.com/ARMmbed/mbed-cloud-client-example"
2. update mbed-os to "mbed-os-6.0.0-beta-1"
3. modify mbed_app.json
    3-1. remove wifi settings
            "nsapi.default-wifi-security"               : "WPA_WPA2",
            "nsapi.default-wifi-ssid"                   : "\"SSID\"",
            "nsapi.default-wifi-password"               : "\"Password\""
    3-2. modify "target.network-default-interface-type" to "CELLULAR"
    3-3. add cellular target			
            "target.macros_add": [
                "CELLULAR_DEVICE=GEMALTO_CINTERION"
            ],
            "target.components_add"                     : ["GEMALTO_CINTERION"],
            "GEMALTO_CINTERION.tx"                      : "PD_5",
            "GEMALTO_CINTERION.rx"                      : "PD_6",
            "GEMALTO_CINTERION.rts"                     : "PD_4",
            "GEMALTO_CINTERION.cts"                     : "PD_3",
            "GEMALTO_CINTERION.provide-default"         : "true",
            "mbed-client-pal.pal-max-frag-len"          : 1,
            "mbed-client.sn-coap-max-blockwise-payload-size": 256,
    3-4. add some settings:
            "mbed-trace.enable"                         : true,
            "cellular.debug-at"                         : true,
            "cellular.use-apn-lookup"                   : false,
            "lwip.ppp-enabled"                          : true,
4. replace mbed_cloud_dev_credentials.c and update_default_resources.c
5. build image and flash it in mcu

Reproduce: 100%

Crash log like below and it keeps crashing and rebooting:

Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20002178, Unknown
Location: 0x805E26B
Error Value: 0x20002178
Current Thread: main Id: 0x20009F10 Entry: 0x805E195 StackSize: 0x1E00 StackMem: 0x2000DA60 SP: 0x2002FF38
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=NUCLEO_F429ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=1) reached maximum, system will halt after rebooting =

jeromecoutant, 2020-05-28:
Hi

> crashed with mbed-os-6.0.0-beta-1 But it works with mbed-os-6.0.0-alpha-3

Ideally, you should find the guilty commit between these 2 tags :-)

> "trace-level": {
> "value": "TRACE_LEVEL_DEBUG"

Could you remove it to check ?

> "cellular.debug-at" : true,

Same check to do

0xc0170, 2020-05-28:
Please do definitely git bisect or at least what has changed for you setup (target code/cellular or something else) ?

dqdqdq31, 2020-05-29:
I found the difference.
I add component SPIF as below without define pins for spif, even though I don't implement or call any SPIF driver, it still crash on mbed-os-6.0.0-beta-1 (but not crash on 6.0.0-alpha-3).
            "target.components_add"                     : ["SPIF"],
            "spif-driver.SPI_FREQ"                      : 40000000

Besides, remove SPIF configs above, it crashed as below:

[INFO][CELL]: AT TX (8): AT^SICS?
Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

[DBG ][CELL]: GEMALTO_CINTERION flow control: RTS 52 CTS 51
[INFO][CELL]: New CellularContext  (0x2000CA50)

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: _device.get_property(AT_CellularDevice::PROPERTY_SOCKET_COUNT) > 0
Location: 0x804E275
File: .\mbed-os\features\cellular\framework\AT\AT_CellularStack.cpp+33
Error Value: 0x0
Current Thread: main Id: 0x20009280 Entry: 0x804FADD StackSize: 0x1400 StackMem: 0x2000A568 SP: 0x2000B788
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=NUCLEO_F429ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=1) reached maximum, system will halt after rebooting =
Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

I only modify mbed_app.json and replace mbed_cloud_dev_credentials.c and update_default_resources.c.
Here's my mbed_app.json

{
    "macros": [
        "ARM_UC_USE_PAL_BLOCKDEVICE=1"
    ],
    "target_overrides": {
        "*": {
            "target.features_add"                       : ["BOOTLOADER", "STORAGE"],
            "platform.stdio-baud-rate"                  : 115200,
            "platform.stdio-convert-newlines"           : true,
            "platform.stdio-buffered-serial"            : true,
            "platform.stdio-flush-at-exit"              : true,
            "rtos.main-thread-stack-size"               : 5120,
            "events.shared-stacksize"                   : 2048,
            "events.shared-eventsize"                   : 2048,
            "update-client.storage-locations"           : 1,
            "mbed-trace.enable"                         : null,
            
            "target.OUTPUT_EXT"                         : "hex"
        },
        "NUCLEO_F429ZI": {
            "target.network-default-interface-type"     : "CELLULAR",
            "target.bootloader_img"                     : "tools/mbed-bootloader-nucleo_f429zi-internal_flash-no_rot-v4.1.0.bin",
            "target.header_offset"                      : "0x8000",
            "target.app_offset"                         : "0x8400",
            "target.restrict_size"                      : "0xF7C00",
            "update-client.bootloader-details"          : "0x080078CC",
            "update-client.application-details"         : "(MBED_ROM_START + MBED_BOOTLOADER_SIZE)",
            "update-client.storage-address"             : "(MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_BASE_ADDRESS+MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_SIZE)",
            "update-client.storage-size"                : "(1024*1024-MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_SIZE)",
            "update-client.storage-page"                : 1,
            "mbed-cloud-client.update-storage"          : "ARM_UCP_FLASHIAP",
            "storage_tdb_internal.internal_base_address": "(MBED_ROM_START+1024*1024)",
            "storage_tdb_internal.internal_size"        : "(128*1024)",
            "storage.storage_type"                      : "TDB_INTERNAL",

            "mbed-trace.enable"                         : true,
            "cellular.debug-at"                         : true,
            "cellular.use-apn-lookup"                   : false,
            "lwip.ppp-enabled"                          : false,

					
            "target.macros_add": [
                "CELLULAR_DEVICE=GEMALTO_CINTERION"
            ],
            "target.components_add"                     : ["GEMALTO_CINTERION"],
            "GEMALTO_CINTERION.tx"                      : "PD_5",
            "GEMALTO_CINTERION.rx"                      : "PD_6",
            "GEMALTO_CINTERION.rts"                     : "PD_4",
            "GEMALTO_CINTERION.cts"                     : "PD_3",
            "GEMALTO_CINTERION.provide-default"         : "true",
            "mbed-client-pal.pal-max-frag-len"          : 1,
            "mbed-client.sn-coap-max-blockwise-payload-size": 256
        }
    },
    "config": {
        "developer-mode": {
            "help"      : "Enable Developer mode to skip Factory enrollment",
            "options"   : [null, 1],
            "value"     : 1
        },
        "button-pinname": {
            "help"      : "PinName for button.",
            "value"     : "BUTTON1"
        },
        "led-pinname"   : {
            "help"      : "PinName for led, which is attached to led blink resource.",
            "value"     : "LED_RED"
        },
        "bootloader-size": {
            "help"      : "Helper macro to enable calculation of rom regions. target.header_offset and target.app_offset still needs to be calculated manually, though.",
            "value"     : "(32*1024)",
            "macro_name": "MBED_BOOTLOADER_SIZE"
        },
        "mbed-cloud-client-update-buffer-size": {
            "value"     : null
        },
        "trace-level": {
            "help": "Options are TRACE_LEVEL_ERROR,TRACE_LEVEL_WARN,TRACE_LEVEL_INFO,TRACE_LEVEL_DEBUG,    TRACE_LEVEL_DEBUG->TRACE_LEVEL_WARN",
            "macro_name": "MBED_TRACE_MAX_LEVEL",
            "value": "TRACE_LEVEL_DEBUG"
        }
    }
}

My build commands:
mbed target NUCLEO_F429ZI
mbed config GCC_ARM_PATH C:\ARM-SDK\gcc-arm-none-eabi-7-2017-q4-major-win32\bin
mbed toolchain GCC_ARM
mbed compile

AriParkkila, 2020-06-02:
Can you share AT log around the line "Cinterion model".

adbridge, 2020-06-12:
@dqdqdq31 please put the issue template back in the header and fill in all the required fields. We reserve the right to close this issue if insufficient information is provided in the correct form. Thank you.

dqdqdq31, 2020-06-15:
What I modified is 
1. mbed_app.json as above
2. main.cpp (for printf log) as below
`// ----------------------------------------------------------------------------
// Copyright 2016-2020 ARM Ltd.
//
// SPDX-License-Identifier: Apache-2.0
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.
// ----------------------------------------------------------------------------

// Needed for PRIu64 on FreeRTOS
#include <stdio.h>
// Note: this macro is needed on armcc to get the the limit macros like UINT16_MAX
#ifndef __STDC_LIMIT_MACROS
#define __STDC_LIMIT_MACROS
#endif

// Note: this macro is needed on armcc to get the the PRI*32 macros
// from inttypes.h in a C++ code.
#ifndef __STDC_FORMAT_MACROS
#define __STDC_FORMAT_MACROS
#endif
#if defined (__RTX)
#include "pdmc_main.h"
#endif
#include "simplem2mclient.h"
#ifdef TARGET_LIKE_MBED
#include "mbed.h"
#endif
#include "application_init.h"
#include "mcc_common_button_and_led.h"
#include "blinky.h"
#ifndef MBED_CONF_MBED_CLOUD_CLIENT_DISABLE_CERTIFICATE_ENROLLMENT
#include "certificate_enrollment_user_cb.h"
#endif

#if defined(MBED_CONF_NANOSTACK_HAL_EVENT_LOOP_USE_MBED_EVENTS) && \
 (MBED_CONF_NANOSTACK_HAL_EVENT_LOOP_USE_MBED_EVENTS == 1) && \
 defined(MBED_CONF_EVENTS_SHARED_DISPATCH_FROM_APPLICATION) && \
 (MBED_CONF_EVENTS_SHARED_DISPATCH_FROM_APPLICATION == 1)
#include "nanostack-event-loop/eventOS_scheduler.h"
#endif

// event based LED blinker, controlled via pattern_resource
#ifndef MCC_MEMORY
static Blinky blinky;
#endif

static void main_application(void);

#if defined(MBED_CLOUD_APPLICATION_NONSTANDARD_ENTRYPOINT)
extern "C"
int mbed_cloud_application_entrypoint(void)
#else
int main(void)
#endif //MBED_CLOUD_APPLICATION_NONSTANDARD_ENTRYPOINT
{
    return mcc_platform_run_program(main_application);
}
[... 387 lines cut ...]
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=NUCLEO_F429ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=89) reached maximum, system will halt after rebooting =
Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20002170, Unknown
Location: 0x8051127
Error Value: 0x20002170
Current Thread: main Id: 0x20009AF0 Entry: 0x8051051 StackSize: 0x1400 StackMem: 0x2000ADD8 SP: 0x2002FF38
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=NUCLEO_F429ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=90) reached maximum, system will halt after rebooting =
Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

pan-, 2020-10-05:
@dqdqdq31 Have you seen any improvements with Mbed OS 6.0 or the latest release ?