ARMmbed/mbed-os #13025: Mbed OS doesn't use latest PSA Crypto API when TF-M is used

Patater, 2020-05-28:
### Description of defect

Mbed OS doesn't use latest PSA Crypto API when TF-M is used.

#### Target(s) affected by this defect ?

All targets using TF-M for their PSA implementation: including PSoC 64 and Musca targets.

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS master after https://github.com/ARMmbed/mbed-os/pull/12955 is merged

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli latest
TF-M 1.0

#### How is this defect reproduced ? 

Attempt to build using PSA Crypto API 1.0 functions that don't yet exist in TF-M's PSA implementation, but do exist in Mbed's PSA Crypto implementation.

Patater, 2020-05-28:
This can by fixed by updating the verion of TF-M that Mbed OS uses to a version of TF-M that has feature parity with the Mbed PSA implementation (which gets pulled in via Mbed TLS), when such a version of TF-M becomes available. TF-M upgrading to get its PSA Crypto implementation from the same version of Mbed TLS that Mbed OS uses will provide such a version of TF-M.

adbridge, 2020-06-15:
@Patater Can you update the header? The description is required to be at least 25 words and the mbed-os version must be a single sha or tag - any other text will cause it to fail the validation!

0xc0170, 2020-06-23:
I edited the description.

adbridge, 2020-06-23:
@Patater Invalid version has been provided so it will not get mirrored until a correct sha/tag is added...

0xc0170, 2020-07-27:
I fixed the sha, taken from PR and updated.

0xc0170, 2020-08-13:
Does this PR fixes it https://github.com/ARMmbed/mbed-os/pull/13418 ?