ARMmbed/mbed-os #13027: CYESKIT_064B0S2_4343W compile error with Mbed 6 and Pelion Client

maclobdell, 2020-05-28:
### Description of defect

Compile error when building the latest `mbed-os-example-pelion` with CYESKIT_064B0S2_4343W. 

Conflict between pinname macro and function pointer typename - each resolving to "A1", etc.

```
./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYESKIT_064B0S2_4343W/cybsp_types.h:232:37: error: expected nested-name-specifier before '(' token
  232 | #define CYBSP_A1                    (P10_1)
      |                                     ^
./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/PinNames.h:145:12: note: in expansion of macro 'CYBSP_A1'
  145 | #define A1 CYBSP_A1
      |            ^~~~~~~~
./mbed-cloud-client/mbed-client/mbed-client/functionpointer.h:113:32: note: in expansion of macro 'A1'
  113 | template <typename R, typename A1>
```

This is resolved by removing parenthesis around pinname definitions in 
`\mbed-os\targets\TARGET_Cypress\TARGET_PSOC6\TARGET_CYESKIT_064B0S2_4343W\cybsp_types.h`  

It looks like the predecessor kit - CY8CKIT_064S2_4343W already has this fix in: 
`\mbed-os\targets\TARGET_Cypress\TARGET_PSOC6\TARGET_CY8CKIT_064S2_4343W\cybsp_types.h`

```
/** Arduino A0 */
#define CYBSP_A0                    P10_0
/** Arduino A1 */
#define CYBSP_A1                    P10_1
/** Arduino A2 */
#define CYBSP_A2                    P10_2
/** Arduino A3 */
#define CYBSP_A3                    P10_3
/** Arduino A4 */
#define CYBSP_A4                    P10_4
/** Arduino A5 */
#define CYBSP_A5                    P10_5
/** Arduino D0 */
#define CYBSP_D0                    (P5_0)

```

#### Target(s) affected by this defect ?

CYESKIT_064B0S2_4343W

#### Toolchain(s) (name and version) displaying this defect ?

Arm compiler 6 

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0-beta-1

mbed-cloud-client 4.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 

- `mbed import https://github.com/ARMmbed/mbed-os-example-pelion`

- copy over this configuration in `mbed_app.json`
```
        "CYESKIT_064B0S2_4343W": {
            "target.macros_add"                         : ["MBEDTLS_USER_CONFIG_FILE=\"mbedTLSConfig_mbedOS.h\"","ARM_UC_USE_PAL_BLOCKDEVICE=0"],
            "target.network-default-interface-type"     : "WIFI",
            "storage_tdb_internal.internal_base_address": "0x101B8000",
            "storage_tdb_internal.internal_size"        : "0x00018000",
            "storage.storage_type"                      : "TDB_INTERNAL",
            "update-client.storage-address"             : "0x100DC000",
            "update-client.storage-size"                : "0x000DC000",
            "update-client.storage-locations"           : 1,
            "mbed-cloud-client.update-storage"          : "ARM_UCP_FLASHIAP",
            "update-client.application-details"         : "0x10000000",
            "target.bootloader_supported"               : true,
            "target.mbed_rom_start"                     : "0x10000400",
            "target.mbed_rom_size"                      : "0x001B7C00",
            "target.sectors"                            : [[268435456, 512]]
        }
```
- copy in new `mbed_cloud_dev_credentials.c`
- `mbed compile -t GCC_ARM -m CYESKIT_064B0S2_4343W --profile release`

kjbracey-arm, 2020-05-29:
I recall hitting this previously - issue was resolved by removing the `()` from the target's pin macro definitions.

(template still works - it gets preprocessed to `template <<typename R, typename P10_1>` which is fine).

Yes, the templates could be _Uglified too, but as the target header defines are injected into any application code including "mbed.h", they're the worse polluter, and making sure that `A1` is defined as a single token makes that intrusion a _bit_ less painful.

kjbracey-arm, 2020-05-29:
Problem recently mentioned in a comment here: https://github.com/ARMmbed/mbed-os/pull/12838#discussion_r412773928

0xc0170, 2020-05-29:
@ARMmbed/team-cypress could you fix the definitions by removing `()` ?

maclobdell, 2020-06-09:
This is fixed by PR #13045. thanks!

closed 2020-06-09 by maclobdell