ARMmbed/mbed-os #13028: STM32WB55 Crypto Hardware Accelaration

felipeLeast, 2020-05-29:
### Description of defect
Hi everyone, I was looking for the stm32wb mbedtls crypto benchmark and then realized that this target don't have the macro MBEDTLS_CONFIG_HW_SUPPORT despite the fact that the datasheet for the device says it should have:
```
– HW public key authority (PKA)
– Cryptographic algorithms: RSA, Diffie-Helman, ECC over GF(p)
– 3x hardware encryption AES maximum 256-bit for the application, the Bluetooth® Low Energy and IEEE802.15.4
– Customer key storage / key manager services
```
Looking in the HAL library seams that all this drivers are present but not being used. I tried to copy the implementation from STM32L4 for mbedtls but some errors emerged and I don't know how to fix them correctly.
```
[Error] aes_alt.c@323,48: 'CRYP_SR_IFEM' undeclared (first use in this function)
[Error] aes_alt.c@323,63: 'CRYP_SR_OFNE' undeclared (first use in this function)
[Error] aes_alt.c@323,78: 'CRYP_SR_BUSY' undeclared (first use in this function); did you mean 'PKA_SR_BUSY'?
[Error] aes_alt.c@329,41: 'CRYP_CR_CRYPEN' undeclared (first use in this function); did you mean 'CRS_CR_CEN'?
[Error] aes_alt.c@330,44: 'AES_TypeDef' {aka 'struct <anonymous>'} has no member named 'IV0LR'
[Error] aes_alt.c@331,44: 'AES_TypeDef' {aka 'struct <anonymous>'} has no member named 'IV0RR'
[Error] aes_alt.c@332,44: 'AES_TypeDef' {aka 'struct <anonymous>'} has no member named 'IV1LR'
[Error] aes_alt.c@333,44: 'AES_TypeDef' {aka 'struct <anonymous>'} has no member named 'IV1RR'

```
If you could help me out I would be glad to create a PR for this.

#### Target(s) affected by this defect ?
NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?
ARM_GCC

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
Don't have a reproduction for this issue

jeromecoutant, 2020-05-29:
Hi
This will be provided with #12747

0xc0170, 2020-05-29:
I will close this as its not an issue (we need to have 12747 in). Let's continue in #12747.

@felipeLeast Please test the pull request

closed 2020-05-29 by 0xc0170

jeromecoutant, 2020-06-08:
Is #12747 working for you ?

felipeLeast, 2020-06-08:
Yes, it is for stm32wb. Only tested with MbedTLS Benchmark so far, but seems to be working.