ARMmbed/mbed-os #13029: Missing ThisThread:: in get_name() of ThisThread.cpp

ndrs-pst, 2020-05-29:
### Description of defect
undefined reference to `rtos::ThisThread::get_name()'
Linking error due to missing of **ThisThread::** in front of get_name()

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 5.15.x

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
call ThisThread::get_name() method

0xc0170, 2020-05-29:
Thanks for the report. How come this was not in our CI? Will investigate

Reference to the declaration: https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/rtos/Thread.h#L397

cc @ARMmbed/mbed-os-core @kjbracey-arm

0xc0170, 2020-05-29:
Pull request welcome to fix this, we might create one soon (in the future for bugs like this it is faster to get the fix in if a pull request is created).

kjbracey-arm, 2020-05-29:
I'm making one.

kjbracey-arm, 2020-05-29:
The odd thing is that it was submitted by someone who said it was useful in #8961. Clearly didn't find it useful enough to actually call it :)

ndrs-pst, 2020-05-29:
@ciarmcom Update already.

0xc0170, 2020-06-03:
sha updated

PR #13030 (2020-06-05): Fix ThisThread::get_name definition

### Summary of changes 

Definition of `ThisThread::get_name` wasn't namespace-qualified, so use would cause a linker error. Correct and test functionality.

Fixes #13029.

#### Impact of changes 

#### Migration actions required 

### Documentation
closed 2020-06-05 by ndrs-pst