ARMmbed/mbed-os #13032: Progress LoRaWAN 1.1 support

martinichka, 2020-05-29:
@ARMmbed/mbed-os-wan
@MathewRJ 

We happily using the LoRaWAN 1.0.2 stack for about two year now, right from the release.

One year ago we read the promising blog from Hasnain Virk that LoRaWAN 1.1 support was implemented. But still today this feature branch did not make it into the master. I understood Hasnain Virk is unfortunately not involved anymore. 

Do you have any expectations on when this might happen? Simple answer like: 'busy with it' or 'it is on hold' would already be nice to know.

Best regards,
Marten Lootsma

0xc0170, 2020-05-29:
Have you already asked on the forum? This is for issue tracker technical issues. I'll close this , we will review however.

closed 2020-05-29 by 0xc0170

0xc0170, 2020-05-29:
The feature has not been finalized (might need some fixes as specs might have been updated over time, the feature has been on a branch for some time). It's on hold, no ETA.

 There has been some fixes from the community, feel free to use it and let us know!