ARMmbed/mbed-os #13049: ThisThread::sleep_for never returns on STM32H743ZI / mbed 6.0.0 beta 1

fabiencomte, 2020-06-01:
Hello,
thread_sleep_for(500) in main function never returns.
Checked with jlink probe / Ozone debugger.

fabiencomte, 2020-06-01:
Hello,
I found that the problem is in rtx_thread.c

`void osRtxThreadListRemove (os_thread_t *thread) {

  if (thread->thread_prev != NULL) {
    thread->thread_prev->thread_next = thread->thread_next;
    if (thread->thread_next != NULL) {
      thread->thread_next->thread_prev = thread->thread_prev; <====== hardfault here, USAGE_FAULT_EXCEPTION reported by mbed_error
    }
    thread->thread_prev = NULL;
  }
}`

fabiencomte, 2020-06-01:
Sometimes thread->thread_next value is 0 (so NULL) but if (thread->thread_next != NULL) go into the if. I don't understand something...

fabiencomte, 2020-06-01:
`  struct osRtxThread_s   *debug_1 = thread->thread_prev;
  if (debug_1 != NULL) {
    thread->thread_prev->thread_next = thread->thread_next;
  struct osRtxThread_s   *debug_2 = thread->thread_next;
    if (debug_2 != NULL) { <= break here
      thread->thread_next->thread_prev = thread->thread_prev;
    }
    thread->thread_prev = NULL;
  }`

when thread->thread_next is 0, debug_2 is not 0 ??? i really not understand. Cache or volatile issue ?

kjbracey-arm, 2020-06-01:
#13050 suggests there are possibly deeper-seated config problems here. I'd want to understand those first.

There's nothing particularly deep or magic in this bit of the RTOS code, assuming the CPU and memory are working correctly. The fact that you can get a fault there suggests it isn't, or there is memory corruption in software somehow.

Please provide toolchain and version details as per the issue template.

fabiencomte, 2020-06-01:
Hello,
Toolchain is Mbed Studio last version.

fabiencomte, 2020-06-01:
Hello,
After migration to 6.0.0 rc1, problem is solved.

fabiencomte, 2020-06-02:
I found why it's gone. My RC1 compilation optimisation is -O1, my beta 1 was -O0. When I changed my RC1 to -O0, the bug returned.

0xc0170, 2020-06-02:
> Please provide toolchain and version details as per the issue template.

@fabiencomte Please fix the first comment (providing issue template with all the required information). It helps all of us to triage this and better understand the problem.

cc @ARMmbed/team-st-mcd

adbridge, 2020-06-12:
@fabiencomte As the requested information has still not been provided (all issues must have all the requested fields filled in) I am closing this. Please re-raise with the required information if this is still an issue.

fabiencomte, 2020-06-13:
@adbridge as in my other report, where are the requested fields ???

fabiencomte, 2020-06-13:
This case is not closed for me. I am ok to give all informations but how ?

kjbracey-arm, 2020-06-15:
When you created the issue, you should have had a template to fill in for the description. Try starting to create a new issue to get a copy of it.

fabiencomte, 2020-06-15:
Ok thank you, I filled it.

adbridge, 2020-06-15:
@fabiencomte thank you for adding the requested information. I've re-opened the issue accordingly :)

fabiencomte, 2020-06-20:
Hello Ciarmcom,
I don't understand your remarks and it looks like a free shotgun hit on me.
- About your 25 words limit. When you run a thread_sleep_for(any ms), it never returns if you use O0 optimisation for compiler, it's very basic. I do not know how to say it with more words. How could you say that better ?
- I have filled all fields as requested by adbridge.

0xc0170, 2020-06-25:
ciarmcom is a bot, the message is auto generated when the template is missing something. I edited the description, should be OK now.

0xc0170, 2020-08-14:
We need to review the ciarcom logic with this issue, it should be without any problems.

cc @ARMmbed/mbed-os-maintainers

LMESTM, 2020-08-14:
@fabiencomte the stack size might be too low when you use -O0.

In mbed, rather than modifying manually the compilers options, user is offered some predefined profiles.
https://os.mbed.com/docs/mbed-os/v6.2/program-setup/build-profiles-and-rules.html

you can the --profile option when building: 
--profile=tools/profiles/debug.json

When debug profile is used, the stack size will be increased automatically: 

$ git grep MBED_CONF_RTOS_IDLE_THREAD_STACK_SIZE_DEBUG_EXTRA                                                                                                                                              rtos/source/TARGET_CORTEX/mbed_rtx_conf.h:#define EXTRA_IDLE_STACK_DEBUG MBED_CONF_RTOS_IDLE_THREAD_STACK_SIZE_DEBUG_EXTRA
--

Because you're not using the profile, I don't think this extra stack size is added ... so this could be a reason for the failure you're seeing ...

fabiencomte, 2020-08-28:
I have found. There was a problem on an interrupt fired during the wait.

closed 2020-08-28 by fabiencomte