ARMmbed/mbed-os #13058: CYESKIT_064B0S2_4343W Test Fails w/ Arm Compiler

maclobdell, 2020-06-03:
### Description of defect

When compiled with Arm Compiler, every Mbed OS test results in `SYNC_FAILED`.

The cause is that the application start address is incorrect. 

**Actual** 
By default, with no configuration changes, the post-build signing script produces an output file with a bootloader header at 0x0FFFFC00 and application at 0x10000000.

**Desired**
The bootloader header should start at 0x10000000 and the application should start at 0x10000400. 

With GCC compiler, it works fine. I suspect a problem with the Arm linker file for this target.

#### Target(s) affected by this defect ?

CYESKIT_064B0S2_4343W 

#### Toolchain(s) (name and version) displaying this defect ?

Arm compiler 9

#### What version of Mbed-os are you using (tag or sha) ?

Tested both `mbed-os-6.0.0-beta-1` and `mbed-os-6.0.0-beta-1`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed cli 1.10.1

#### How is this defect reproduced ? 
get a board, provision it
import mbed os
run tests
` mbed test -m CYESKIT_064B0S2_4343w -t ARM`
view the output hex files

maclobdell, 2020-06-03:
@sreeharshaangara @kyle-cypress can you check this. Please confirm test results on your side with the arm compiler.

maclobdell, 2020-06-03:
I was able to workaround this bug by making the following modifications to the linker script.

`#define FLASH_START             MBED_APP_START+BOOT_HEADER_SIZE`

```
LR_IROM1 FLASH_START FLASH_SIZE
{
    ER_FLASH_VECTORS +0
    {
        * (RESET, +FIRST)
    }
```

vmedcy, 2020-06-11:
Hi @maclobdell  - thanks for reporting this bug. I investigated the behavior of ARM linker, fromelf tool and confirm the linker script tweak should address the issue (LR_IROM1 should start from 0x10000400 not 0x10000000 to produce the correct unsigned HEX).

I pushed https://github.com/ARMmbed/mbed-os/pull/13100 PR.

maclobdell, 2020-06-11:
@vmedcy Thanks!

closed 2020-06-16 by 0xc0170