ARMmbed/mbed-os #13059: Platform dependent enum type may cause problems

WisniewskiP, 2020-06-03:
### Description of defect

Short description:
arm-none-eabi-gcc uses unigned char as underlying type for enum when type is not fixed.
In some error cases -1 is assigned as return value. This causes truncation and instead of expected -1, 255 is assigned.
You can observe one of this cases in:
TEST_F(TestAT_CellularNetwork, test_AT_CellularNetwork_get_network_registering_mode) line 366
cn.get_network_registering_mode(mode) is returning error but mode is set to 0xFF and not to -1 as expected on next line.

It works on x86 because gcc, for this platform, uses unsigned int for this enum.

Long one:
https://embeff.com/mbed-unit-tests-investigation/

there are probably also other places with similar problems.

#### Target(s) affected by this defect ?
All targets build with arm-none-eabi-gcc

#### Toolchain(s) (nme and version) displaying this defect ?
arm-none-eabi-gcc 8.2, 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?

SHA: 098c72a31277e0b292dd6cdf1abaf9ca9d487032

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
build Test TEST_F(TestAT_CellularNetwork, test_AT_CellularNetwork_get_network_registering_mode) from at_cellularnetworktest.cpp with arm-none-eabi-gcc and run on target.

0xc0170, 2020-06-03:
Thanks for the report ! Worth readin the report referenced above.

cc @ARMmbed/mbed-os-wan

kjbracey-arm, 2020-06-03:
Pedantry time :) From your article, you say

> That means that during assignment int will be converted to unsigned char (or unsigned int in the original case on our host system). 

It's actually a bit more subtle than that. C++14 said

> A value of integral or enumeration type can be explicitly converted to an enumeration type. The value is unchanged if the original value is within the range of the enumeration values (7.2). Otherwise, the resulting value is unspecified (and might not be in that range).

All out-of-range assignments were unspecified.

C++20 draft says

> A value of integral or enumeration type can be explicitly converted to a complete enumeration type. If the enumeration type has a fixed underlying type, the value is first converted to that type by integral conversion, if necessary, and then to the enumeration type. If the enumeration type does not have a fixed underlying type, the value is unchanged if the original value is within the range of the enumeration values (9.7.1), and otherwise, the behavior is undefined.

So newer versions have defined the behaviour for out-of-range assignments for fixed types, but reduced non-fixed-type enums to be totally undefined.

Not checked what the current C rules are.

kivaisan, 2020-06-03:
Good catch. In this particular UT failure, the whole `EXPECT_TRUE(mode == -1);` is an invalid check. If getter returns an error, you cannot rely on `mode` value. I'll make a fix PR to remove it.

AriParkkila, 2020-06-05:
@WisniewskiP please see #13069.

PR #13069 (2020-06-10): Cellular: Fix unittest for CellularNetwork

### Summary of changes 

Fix #13059.

#### Impact of changes 

#### Migration actions required
closed 2020-06-10 by 0xc0170