ARMmbed/mbed-os #13061: Function get_erase_size is declared incorrectly in QSPIFBlockDevice.h

jporter4, 2020-06-03:
Function get_erase_size is declared incorrectly in QSPIFBlockDevice.h.  It is missing the keyword 'const'.

Tested with external Flash device on Nordic nRF52840 based project.  I would expect to see the same warning with any external QSPIF device.

5.14.1 and 5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
EWARM 8.50

#### How is this defect reproduced ? 
Include the file and compile. 
It generates a warning only, so low severity.

0xc0170, 2020-06-05:
Thanks for the report. @jporter4 Please send PR fixing it.

As it was tested on 5.15, is it also present on master?

jporter4, 2020-06-05:
Hi,
If you’re asking if the ‘const’ keyword is still missing in version 6.0.0 – yes, it is still missing:

QSPIFBlockDevice.h:
virtual mbed::bd_size_t get_erase_size(mbed::bd_addr_t addr);

should be:
virtual mbed::bd_size_t get_erase_size(mbed::bd_addr_t addr) const;

Confidentiality Notice: This electronic mail transmission is intended for the use of the individual or entity to which it is addressed and may contain confidential and/or proprietary information belonging to the sender. If you are not the intended recipient, you are hereby notified that any disclosure, use, copying, distribution, or the taking of any action in reliance on the contents of this information is strictly prohibited. If you have received this transmission in error, please notify the sender immediately by e-mail and delete the original message. Thank you for your cooperation.

adbridge, 2020-06-12:
@jporter4 when raising issues please do not delete sections from the template. All sections are required to be filled in. Thanks. We reserve the right to reject issues that do not contain sufficient information for us to reproduce correctly.

jporter4, 2020-06-12:
If I deleted any sections, it was accidental.
What information are you missing that you need to confirm a syntax error in a source file ?

0xc0170, 2020-06-17:
@jporter4 thanks for the confirmation. Can you send a pull request addressing this ?

PR #13960 (2020-11-26): Add const keyword to get_erase_size declaration

### Summary of changes 
Fixes #13061
Const keyword added to declaration of get_erase_size. Const objects are only able to access const methods, without it a compiler error is raised when a const object attempts to use a non-const method. This change fixes this and removes related compiler warnings. 

#### Impact of changes 

None
#### Migratio
closed 2020-11-26 by 0xc0170