ARMmbed/mbed-os #13062: mbed-os-example-mesh-minimal build fails with KW41Z and GCC_ARM

artokin, 2020-06-03:
### Description of defect

mbed-os-example-mesh-minimal fails to build against Mbed OS master when using command:

`mbed compile -t GCC_ARM -m KW41Z --app-config ./configs/mesh_6lowpan.json`

The build produces error:
```
Link: mbed-os-example-mesh-minimal
BUILD/KW41Z/GCC_ARM/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/fsl_common.o: In function `InstallIRQHandler':
/home/arto/github/mbed-os-example-mesh-minimal/./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/fsl_common.c:63: undefined reference to `__Vectors'
collect2: error: ld returned 1 exit status
[ERROR] BUILD/KW41Z/GCC_ARM/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/fsl_common.o: In function `InstallIRQHandler':
/home/arto/github/mbed-os-example-mesh-minimal/./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/fsl_common.c:63: undefined reference to `__Vectors'
collect2: error: ld returned 1 exit status

```

The application builds successfully with ARM toolchain.

#### Target(s) affected by this defect ?
KW41Z

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

165be79392ae7b1bee4388d2bc8ed8281978f07c

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

- git clone git@github.com:ARMmbed/mbed-os-example-mesh-minimal.git
- cd mbed-os-example-mesh-minimal
- mbed deploy
- cd mbed-os
- git checkout master
- cd ..
- mbed compile -t GCC_ARM -m KW41Z --app-config ./configs/mesh_6lowpan.json

MarceloSalazar, 2020-06-03:
Note this seems a regression introduced in Mbed 6.
Mbed OS 5.15.3 compiles fine with GCC_ARM.

0xc0170, 2020-06-04:
I fixed the SHA. we need to fix the script.

This PR broke it https://github.com/ARMmbed/mbed-os/pull/12949, updating CMSIS files. I fixed some breakages we found but this one was not caught. The simple example/tests passed for this target. 🔥 

This comes from CMSIS files, there is:

```
#ifndef __VECTOR_TABLE
#define __VECTOR_TABLE            __Vectors
#endif
```
Using `__name` in CMSIS common header and not expecting a clash with other already supported CMSIS MCUs 🙄  As this symbol is only for GCC ARM, it passes with ARMCC or IAR.

I'll look at this  tomorrow how to fix it. Thanks for the report.

0xc0170, 2020-06-04:
Fix proposed in #13064

0xc0170, 2020-06-04:
I was able to reproduce and the fix should resolve this.

PR #13064 (2020-06-04): nxp: fix vector redefinition from CMSIS

### Summary of changes 

Since 5.7 CMSIS update to Mbed OS, __VECTOR_TABLE is defined in cmsis_gcc header
file. Many MCU in NXP uses this symbol as linker definition, therefore we should
check if already defined and undefined it.

Fixes #13062

#### Impact of changes 

#### Migration actions required
commit 41995a74 (2020-06-04): nxp: fix vector redefinition from CMSIS

Since 5.7 CMSIS update to Mbed OS, __VECTOR_TABLE is defined in cmsis_gcc header
file. Many MCU in NXP uses this symbol as linker definition, therefore we should
check if already defined and undefined it.

Fixes #13062
closed 2020-06-04 by 0xc0170