ARMmbed/mbed-os #13065: open62541 library implementation in mbed Project on K64F

ThePhilomathsApprentice, 2020-06-04:
### Description of defect
This is a Question.
I want to implement open62541 library in mbed OS project on K64F board.
I want to know whether this is possible or not. if yes, Is there some online documentation related to the same.

#### Target(s) affected by this defect ?
-K64F

#### Toolchain(s) (name and version) displaying this defect ?
-GCC, Mbed-cli

#### What version of Mbed-os are you using (tag or sha) ?
-mbedOS v5.11.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
-mbed-cli

#### How is this defect reproduced ?

0xc0170, 2020-06-04:
Please use forum.mbed.com for questions. I'll close this question.

closed 2020-06-04 by 0xc0170

ThePhilomathsApprentice, 2020-06-09:
How many mbed OS version 5.11.4 do you have?