ARMmbed/mbed-os #13068: mbed-os-6.0.0  build error

fengshengchang, 2020-06-05:
### Description of defect
**I updated os to mbed-os-6.0.0 and build my application. I got the following error. Please tell me how to solve it** 

G:\nbproject>mbed compile -t GCC_ARM -m NUCLEO_F103RB --source nbapp --source mbed-os-6.0.0 --build BUILD/nbapp/release -D NDEBUG -v & xcopy .\BUILD\nbapp\release\nbapp.bin .\BUILD\nbapp\release\nbapp_release.bin /y
[mbed] Working path "G:\nbproject" (program)
[mbed] Exec "d:\python27\python.exe -u G:\nbproject\mbed-os\tools\make.py -D NDEBUG -t GCC_ARM -m NUCLEO_F103RB --source nbapp --source mbed-os-6.0.0 --build BUILD/nbapp/release -v" in "G:\nbproject"
Building project nbapp (NUCLEO_F103RB, GCC_ARM)
Scan: nbapp
Scan: mbed-os-6.0.0
Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DTARGET_STM32F1 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_NAME=NUCLEO_F103RB -DMBED_BUILD_TIMESTAMP=1591327334.73 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DCOMPONENT_PSA_SRV_EMUL=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_M3 -DDEVICE_PWMOUT=1 -DDEVICE_SPI_ASYNCH=1 -DUSE_FULL_LL_DRIVER -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DUSE_HAL_DRIVER -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DTOOLCHAIN_GCC_ARM -DDEVICE_SERIAL_FC=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTARGET_NUCLEO_F103RB -DDEVICE_RESET_REASON=1 -DDEVICE_SPI=1 -DCOMPONENT_NSPE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DNDEBUG -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_STM32F103RB
Compile [  0.5%]: at24mac.cpp
Compile: arm-none-eabi-g++ -std=gnu++14 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x20000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x5000 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DTARGET_STM32F1 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_NAME=NUCLEO_F103RB -DMBED_BUILD_TIMESTAMP=1591327334.73 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DCOMPONENT_PSA_SRV_EMUL=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_M3 -DDEVICE_PWMOUT=1 -DDEVICE_SPI_ASYNCH=1 -DUSE_FULL_LL_DRIVER -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DUSE_HAL_DRIVER -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DTOOLCHAIN_GCC_ARM -DDEVICE_SERIAL_FC=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTARGET_NUCLEO_F103RB -DDEVICE_RESET_REASON=1 -DDEVICE_SPI=1 -DCOMPONENT_NSPE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DNDEBUG -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_STM32F103RB @BUILD/nbapp/release\.includes_72997ad1fb063e7065c1fd9f7c7d489f.txt -include BUILD/nbapp/release\mbed_config.h -MD -MF BUILD\nbapp\release\mbed-os-6.0.0\components\802.15.4_RF\atmel-rf-driver\source\at24mac.d -o BUILD\nbapp\release\mbed-os-6.0.0\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o mbed-os-6.0.0\components\802.15.4_RF\atmel-rf-driver\source\at24mac.cpp
[Error] I2C.h@108,15: 'i2c_pinmap_t' does not name a type
[Error] I2C.h@109,15: 'i2c_pinmap_t' does not name a type
[DEBUG] Return: 1
[DEBUG] Output: In file included from mbed-os-6.0.0\components\802.15.4_RF\atmel-rf-driver\source\at24mac.h:23:0,
[DEBUG] Output:                  from mbed-os-6.0.0\components\802.15.4_RF\atmel-rf-driver\source\at24mac.cpp:16:
[DEBUG] Output: mbed-os-6.0.0/drivers/I2C.h:108:15: error: 'i2c_pinmap_t' does not name a type
[DEBUG] Output:      I2C(const i2c_pinmap_t &static_pinmap);
[DEBUG] Output:                ^~~~~~~~~~~~
[DEBUG] Output: mbed-os-6.0.0/drivers/I2C.h:109:15: error: 'i2c_pinmap_t' does not name a type
[DEBUG] Output:      I2C(const i2c_pinmap_t &&) = delete; // prevent passing of temporary objects
[DEBUG] Output:                ^~~~~~~~~~~~
Traceback (most recent call last):
  File "G:\nbproject\mbed-os\tools\make.py", line 80, in wrapped_build_project

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

fengshengchang, 2020-06-05:
1、version of Mbed OS：
[image]
2、toolchain is GCC_ARM  and the target is NUCLEO_F103RB
[image]
3、My Application test code is very simple, just a main function:
[image]
4、mbed cli version as follow:
[image]

0xc0170, 2020-06-05:
First, you need to provide more details about the project - what are you building, this is not an example that we could get?

Second, `mbed compile -t GCC_ARM -m NUCLEO_F103RB` is not sufficient, you specify `--sources` ?

`i2c_pinmap_t` comes from i2c_api header. I see you got `-DDEVICE_I2C` in there, but is it getting pulled in and included? Just verify.

If you reported 6.0.0 issue, was this working with any previous version?

fengshengchang, 2020-06-08:
It was working on mbed-os-5.13 and I am sure i2c_api header is getting pulled in mbed-os-6.0.0 . 
Can you please just try the following complie command, this action can also cause errors:
mbed compile -t GCC_ARM -m NUCLEO_F103RB **--source ts** --source mbed-os-6.0.0

My ts directory just have one main.cpp file, and the content of main.cpp is as follows:

#include "mbed.h"
int main()
{
	debug("\n\n*** main thread end***\n");
}

0xc0170, 2020-06-08:
@ARMmbed/team-st-mcd Is this related anyhow to the baremetal changes ? This target supports only baremetal ?

fengshengchang, 2020-06-08:
It was working on mbed-os-5.13 and mbed-os-5.14.
But when I try mbed-os-5.15, I got the same errors on mbed-os-5.15.

0xc0170, 2020-06-09:
@fengshengchang would you be able `git bisect` to find what commit caused it ? It might be easier on 5.15 branch.

adbridge, 2020-06-15:
@fengshengchang please update the issue header with the requested information so that we may proceed with this issue.

adbridge, 2020-06-23:
@fengshengchang as the issue header has still not been updated with the requested information after 18 days we are closing this. Please re-open once the required information has been added. Thank you.

closed 2020-06-23 by adbridge