ARMmbed/mbed-os #13074: CYESKIT_064B0S2_4343W UnicodeEncodeError: 'charmap' codec can't encode characters

maclobdell, 2020-06-06:
### Description of defect

After applying work-arounds for https://github.com/ARMmbed/mbed-os/issues/13058 and https://github.com/ARMmbed/mbed-os/issues/13027, I ran mbed os tests. 

Several tests result in an UnicodeEncodeError and the results are not conclusive. In the test summary they show as SKIPPED. 

See full test summary:
[CYESKIT_064B0S2_4343W_GCC_ARM_test_results.txt](https://github.com/ARMmbed/mbed-os/files/4739061/CYESKIT_064B0S2_4343W_GCC_ARM_test_results.txt)

When I run a single test with verbose output, I can see the `UnicodeEncodeError: 'charmap' codec can't encode characters` in the output. Such as in this log:
[cyeskit-test-events-queue.txt](https://github.com/ARMmbed/mbed-os/files/4739081/cyeskit-test-events-queue.txt)

#### Target(s) affected by this defect ?

CYESKIT_064B0S2_4343W 

#### Toolchain(s) (name and version) displaying this defect ?

GCC Arm compiler 9

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.0.0-rc-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli           1.10.1
mbed-cloud-sdk     2.0.8
mbed-flasher       0.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.10
mbed-os-tools      0.0.12
[full list of python packages](https://github.com/ARMmbed/mbed-os/files/4739072/pip_list.txt)

#### How is this defect reproduced ? 
-get a board, provision it
-import mbed os
-apply workarounds as noted in above issues
-run a single test suite
` mbed test -m CYESKIT_064B0S2_4343W -t GCC_ARM -n tests-events-queue -v`
-view the terminal output

0xc0170, 2020-06-08:
cc @ARMmbed/team-cypress

adbridge, 2020-06-15:
@maclobdell I've corrected 'mbed-os-6.0.0-rc-1' to 'mbed-os-6.0.0-rc1' to get the right tag!

maclobdell, 2020-07-17:
tested production kit. I don't see this issue any longer. suspect there was a mis-configuration in the clock for the engineering sample kit, but it is resolved now on the production kit that superseded it.

closed 2020-07-17 by maclobdell