ARMmbed/mbed-os #13075: CYESKIT_064B0S2_4343W Mbed OS 6 Test fails

maclobdell, 2020-06-06:
### Description of defect

After applying work-arounds for https://github.com/ARMmbed/mbed-os/issues/13058 and https://github.com/ARMmbed/mbed-os/issues/13027, I ran mbed os tests. 

The following test result in FAIL or TIMEOUT:

features-feature_ble-targets-target_cordio-tests-cordio_hci-driver           | FAIL    | 57.14
tests-mbed_drivers-reset_reason                                              | TIMEOUT | 130.08            
tests-mbed_drivers-timeout                                                   | FAIL    | 41.08     
tests-mbed_hal-reset_reason                                                  | TIMEOUT | 130.03  
tests-mbed_hal-rtc_reset                                                     | TIMEOUT | 154.11  

See full test summary:
[CYESKIT_064B0S2_4343W_GCC_ARM_test_results.txt](https://github.com/ARMmbed/mbed-os/files/4739061/CYESKIT_064B0S2_4343W_GCC_ARM_test_results.txt)

#### Target(s) affected by this defect ?

CYESKIT_064B0S2_4343W 

#### Toolchain(s) (name and version) displaying this defect ?

GCC Arm compiler 9

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.0.0-rc-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli           1.10.1
mbed-cloud-sdk     2.0.8
mbed-flasher       0.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.10
mbed-os-tools      0.0.12
[full list of python packages](https://github.com/ARMmbed/mbed-os/files/4739072/pip_list.txt)

#### How is this defect reproduced ? 
-get a board, provision it
-import mbed os
-apply workarounds as noted in above issues
-run tests
` mbed test -m CYESKIT_064B0S2_4343W -t GCC_ARM`
-view the terminal output

0xc0170, 2020-06-08:
cc @ARMmbed/team-cypress

maclobdell, 2020-06-15:
I updated the description with the commit hash instead of the tag name. It appears the automatic auto parsing is not working with the tag mbed-os-6.0.0-rc1

maclobdell, 2020-07-17:
Tested production kit. everything but the cordio test passes now. that is covered by #13295. closing this one.

closed 2020-07-17 by maclobdell