ARMmbed/mbed-os #13076: CY8CKIT_062S2_43012 Test Fails

maclobdell, 2020-06-06:
### Description of defect

The following test result in FAIL with ` CY8CKIT_062S2_43012` and `Mbed OS 6.0`.
```
| CY8CKIT_062S2_43012-ARMC6   | tests-mbed_hal-sleep                                                         | FAIL   |  16.98 |
| CY8CKIT_062S2_43012-ARMC6   | tests-mbed_hal-sleep_manager                                                 | FAIL   |  15.33 |
| CY8CKIT_062S2_43012-ARMC6   | features-feature_ble-targets-target_cordio-tests-cordio_hci-driver           | FAIL   |  34.17 |
| CY8CKIT_062S2_43012-GCC_ARM | tests-mbed_hal-sleep_manager                                                 | FAIL   |  15.46 |
| CY8CKIT_062S2_43012-GCC_ARM | tests-mbed_hal-sleep                                                         | FAIL   |  16.99 |
| CY8CKIT_062S2_43012-GCC_ARM | features-feature_ble-targets-target_cordio-tests-cordio_hci-driver           | FAIL   |  34.16 |
```

See full test summaries:
[CY8CKIT_062S2_43012_GCC_ARM_results.txt](https://github.com/ARMmbed/mbed-os/files/4739238/CY8CKIT_062S2_43012_GCC_ARM_results.txt)
[CY8CKIT_062S2_43012_ARM_results.txt](https://github.com/ARMmbed/mbed-os/files/4739239/CY8CKIT_062S2_43012_ARM_results.txt)

#### Target(s) affected by this defect ?

CY8CKIT_062S2_43012

#### Toolchain(s) (name and version) displaying this defect ?

GCC Arm compiler 9
Arm compiler 6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.0.0-rc-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli           1.10.1
mbed-cloud-sdk     2.0.8
mbed-flasher       0.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.10
mbed-os-tools      0.0.12
[full list of python packages](https://github.com/ARMmbed/mbed-os/files/4739072/pip_list.txt)

#### How is this defect reproduced ? 
```
mbed import https://github.com/armmbed/mbed-os
mbed test -m CY8CKIT_062S2_43012 -t GCC_ARM
mbed test -m CY8CKIT_062S2_43012 -t ARM
```

0xc0170, 2020-06-08:
cc @ARMmbed/team-cypress

kyle-cypress, 2020-06-08:
Not sure about the cordio failures yet, but the sleep failures are probably the same as #12434 (known issue, fix in progress).

maclobdell, 2020-07-21:
I updated the tag to `mbed-os-6.0.0`. Hopefully that gets the script off my case.

adbridge, 2020-11-17:
> 
> 
> Not sure about the cordio failures yet, but the sleep failures are probably the same as #12434 (known issue, fix in progress).

@kyle-cypress as #12434 is now fixed can this be closed ?

kyle-cypress, 2020-11-17:
Yes, I believe this can be closed. The Cordio failure can be tracked in #13295.

PR #14982 (2021-08-13): CYW43XXX Cordio HCI driver: fixed MCU deep-sleep locking flow

### Summary of changes 
Fixed MCU deep-sleep locking flow in CYW43XXX Cordio HCI driver.  The CYW43XXX Cordio HCI driver incorrect handles lock/unlock deep-sleep entry, as results it allows to put PSoC6 in to deep-sleep when CYW43xx does not expect this (the host_wake_up pin in ACTIVE_LOW state, which means hat host must be in ACTIVE mode ). 

Now, CYW43XXX Cordio HCI driver has the following MCU deep-sleep locking scenarios/flows:
a) A BT device or MCU does not support Low Power mode (MBED configuration does not include MBED_TICKL [... cut]
commit 1d6843e5 (2021-08-13): Fixed MCU deep-sleep locking flow in CYW43XXX Cordio HCI driver. The CYW43XXX Cordio HCI driver incorrect handles lock/unlock deep-sleep entry, as results it allows to put PSoC6 in to deep-sleep when CYW43xx does not expect this (the host_wake_up pin in ACTIVE_LOW state, which means hat host must be in ACTIVE mode ).

Now, CYW43XXX Cordio HCI driver has the following MCU deep-sleep locking scenarios/flows:
a) A BT device or MCU does not support Low Power mode (MBED configuration does not include MBED_TICKLESS, DEVICE_SLEEP, DEVICE_LPTICKER or CYCFG_BT_LP_ENABLED features). In this case, CyH4Tr [... cut]
closed 2020-11-17 by 0xc0170