ARMmbed/mbed-os #13078: std::cout does not print out variables

bluePain, 2020-06-07:
### Description of defect

std::cout does not print out variables over USB serial as in v5.15.3

#### Target(s) affected by this defect ?

IMXRT1050-EVKB

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio: 0.9.1

#### How is this defect reproduced ? 

Compiling the following code piece.

`std::cout << "Thread ID: " << ThisThread::get_id() << std::endl;`

v6.0.0 output:

    Thread ID:

v5.15.3 output:

    ThreadID: 80808080

kjbracey-arm, 2020-06-08:
Does setting the config option `target.printf_lib` to `"std"` solve this?

bluePain, 2020-06-08:
That actually fixed the issue. Sorry for bothering. 

However (and to be honest) I couldn't figure out how to follow the API guideline to find out the solution. Even in the main documentation, it was not clear I suppose (talking about [this page](https://os.mbed.com/docs/mbed-os/v5.15/tutorials/debugging-using-printf-statements.html)) The only place I found relevant information is [here](https://github.com/ARMmbed/mbed-os/blob/master/platform/source/minimal-printf/README.md)

Maybe I just don't know how to read at all :). Whatever, thanks anyway.

closed 2020-06-08 by bluePain