ARMmbed/mbed-os #13094: K64F Hard fault in EMAC startup

artokin, 2020-06-10:
### Description of defect

https://github.com/ARMmbed/nanostack-border-router application compiled against Mbed OS master is making a hard fault at start up. It looks like EMAC driver initialisation fails. 

The hard fault happens on default profile, it is not observed with debug profile

```
[INFO][AtRF]: RF part number: b
[INFO][app]: Starting NanoStack Border Router...
[INFO][app]: Build date: Jun 10 2020 10:25:46
[INFO][app]: Mbed OS: 60000
[INFO][brro]: NET_IPV6_BOOTSTRAP_AUTONOMOUS
[INFO][app]: Using EMAC backhaul driver...

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0: 30000858
R1: 20000590
R2: 100002C8
R3: 0
R4: 20002C7C
R5: 20002CD8
R6: 20002CC0
R7: 0
R8: FFFFFFFF
R9: 45
R10: 1
R11: 0
R12: 0
SP : 20002C40
LR : 4D57
PC : 0
xPSR : 40000000
PSP : 20002C20
MSP : 2002FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 0
BFSR : 0
UFSR : 2
DFSR : 8
AFSR : 0
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x0
Error Value: 0x1FFF0400
Current Thread: shared_event_queue Id: 0x20002434 Entry: 0x59E81 StackSize: 0x800 StackMem: 0x20002478 SP: 0x20002C40
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=K64F
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=2) reached maximum, system will halt after rebooting =

```

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

998d06a80b185311ecdd4feac230aa8737444af5

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
1. clone https://github.com/ARMmbed/nanostack-border-router
2. cd nanostack-border-router
3. mbed deploy
4. cd mbed-os
5. git checkout master
6. cd ..
7.  mbed compile -t GCC_ARM -m K64F --app-config ./configs/Thread_Atmel_RF.json 
8. Flash the application to K64F with Firefly RF shield
9. Observe the result

0xc0170, 2020-06-10:
Do we know when it started failing? I dont recall any k64f fixes, my first suspect would be CMSIS update that caused vector redefinition. This one #13064 or one earlier that brought in CMSIS 5.7 ?

adbridge, 2020-06-15:
@artokin You need to specify tool versions or n/a

closed 2021-10-20 by ciarmcom