ARMmbed/mbed-os #13099: Some Mbed OS 6 targets don't compile with Baremetal profile

MarceloSalazar, 2020-06-11:
# Description of defect

Mbed OS 6 has enabled the bare-metal profile for a few targets, but not all of them (details in targets.json)

Developers are trying to use the [mbed-os-example-blinky-baremetal](https://github.com/ARMmbed/mbed-os-example-blinky-baremetal) application with some Mbed OS 6 targets that are not yet supported - see the issue raised in forums: https://forums.mbed.com/t/some-issues-with-the-mbedos6-and-mbed-tools

Eventually **all** targets should support the bare-metal profile as a minimum requirement.
Until then, this issue with list of targets should help to keep track of the status.

#### Target(s) affected by this defect ?

Targets marked with :x: fail to compile.

| Target	|GCC|ARM|
| --------------|---------------|---------------|
|KW41Z         |:x:|:x:|
|CY8CKIT_062_BLE    |:x:|:x:|
|UHURU_RAVEN      |:heavy_check_mark:|:heavy_check_mark:|
|RZ_A1H        |:heavy_check_mark:|:heavy_check_mark:|
|ARM_MPS2_M0      |:heavy_check_mark:|:x:|
|ARM_MPS2_M3      |:heavy_check_mark:|:x:|
|ARM_MPS2_M4      |:heavy_check_mark:|:x:|
|ARM_MPS2_M7      |:heavy_check_mark:|:x:|
|K66F          |:heavy_check_mark:|:heavy_check_mark:|
|MAX32600MBED     |:heavy_check_mark:|:x:|
|FF_LPC546XX      |:heavy_check_mark:|:x:|
|NUCLEO_L496ZG_P    |:heavy_check_mark:|:x:|
|CY8CPROTO_062_4343W  |:x:|:x:|
|NUCLEO_F207ZG     |:heavy_check_mark:|:heavy_check_mark:|
|DISCO_F746NG     |:heavy_check_mark:|:heavy_check_mark:|
|CYW9P62S1_43012EVB_01 |:x:|:x:|
|NUCLEO_F411RE     |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F413ZH     |:heavy_check_mark:|:x:|
|ARCH_PRO       |:heavy_check_mark:|:heavy_check_mark:|
|MAX32625PICO     |:heavy_check_mark:|:x:|
|NUCLEO_L433RC_P    |:heavy_check_mark:|:x:|
|ARCH_MAX       |:heavy_check_mark:|:x:|
|ADV_WISE_1510     |:heavy_check_mark:|:heavy_check_mark:|
|DISCO_L562QE     |:heavy_check_mark:|:heavy_check_mark:|
|RHOMBIO_L476DMW1K   |:heavy_check_mark:|:x:|
|KL43Z         |:heavy_check_mark:|:heavy_check_mark:|
|K22F         |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_L432KC     |:heavy_check_mark:|:x:|
|SDP_K1        |:heavy_check_mark:|:x:|
|NUCLEO_F303ZE     |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_L552ZE_Q    |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F401RE     |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F303RE     |:heavy_check_mark:|:heavy_check_mark:|
|MOTE_L152RC      |:heavy_check_mark:|:x:|
|FVP_MPS2_M3      |:heavy_check_mark:|:x:|
|FVP_MPS2_M0      |:heavy_check_mark:|:x:|
|DISCO_L476VG     |:heavy_check_mark:|:x:|
|FVP_MPS2_M7      |:heavy_check_mark:|:x:|
|FVP_MPS2_M4      |:heavy_check_mark:|:x:|
|ARM_MUSCA_A1_NS    |:x:|:x:|
|NUCLEO_F429ZI     |:heavy_check_mark:|:heavy_check_mark:|
|NU_PFM_M2351_NPSA_S  |:x:|:x:|
|NUMAKER_PFM_M453   |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F091RC     |:heavy_check_mark:|:x:|
|SDT52832B       |:heavy_check_mark:|:x:|
|LPC546XX       |:heavy_check_mark:|:x:|
|MAX32630FTHR     |:heavy_check_mark:|:x:|
|TMPM46B        |:heavy_check_mark:|:x:|
|DISCO_L496AG     |:heavy_check_mark:|:x:|
|FVP_MPS2_M0P     |:heavy_check_mark:|:x:|
|NUMAKER_PFM_NUC472  |:heavy_check_mark:|:heavy_check_mark:|
|ARM_MUSCA_B1_NS    |:x:|:x:|
|DISCO_F769NI     |:heavy_check_mark:|:heavy_check_mark:|
|GD32_F307VG      |:heavy_check_mark:|:x:|
|EV_COG_AD4050LZ    |:heavy_check_mark:|:heavy_check_mark:|
|DISCO_H747I      |:heavy_check_mark:|:x:|
|SDT64B        |:heavy_check_mark:|:heavy_check_mark:|
|XDOT_L151CC      |:heavy_check_mark:|:x:|
|NUCLEO_L486RG     |:heavy_check_mark:|:x:|
|CY8CKIT_062_WIFI_BT  |:x:|:x:|
|SDT32620B       |:heavy_check_mark:|:x:|
|GR_LYCHEE       |:heavy_check_mark:|:heavy_check_mark:|
|MAX32620FTHR     |:heavy_check_mark:|:x:|
|NUCLEO_G474RE     |:x:|:x:|
|GD32_F450ZI      |:heavy_check_mark:|:x:|
|ARM_CM3DS_MPS2    |:heavy_check_mark:|:x:|
|NUCLEO_L476RG     |:heavy_check_mark:|:x:|
|S5JS100        |:x:|:x:|
|NUCLEO_F746ZG     |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_WB55RG     |:heavy_check_mark:|:x:|
|MTS_DRAGONFLY_F413RH |:heavy_check_mark:|:x:|
|DISCO_L4R9I      |:heavy_check_mark:|:x:|
|GENERIC_H745I_CM4   |:heavy_check_mark:|:x:|
|GENERIC_H745I_CM7   |:heavy_check_mark:|:x:|
|NUMAKER_M252KG    |:heavy_check_mark:|:heavy_check_mark:|
|DISCO_F413ZH     |:heavy_check_mark:|:x:|
|NUCLEO_F756ZG     |:heavy_check_mark:|:heavy_check_mark:|
|NUMAKER_PFM_NANO130  |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_L496ZG     |:heavy_check_mark:|:x:|
|CY8CPROTO_064_SB   |:x:|:x:|
|NUCLEO_L152RE     |:heavy_check_mark:|:x:|
|NUCLEO_F439ZI     |:heavy_check_mark:|:x:|
|EP_AGORA       |:heavy_check_mark:|:heavy_check_mark:|
|DISCO_F469NI     |:heavy_check_mark:|:x:|
|FF1705_L151CC     |:heavy_check_mark:|:x:|
|K82F         |:heavy_check_mark:|:heavy_check_mark:|
|CYESKIT_064B0S2_4343W |:x:|:x:|
|TMPM4G9        |:heavy_check_mark:|:x:|
|EFM32GG11_STK3701   |:heavy_check_mark:|:x:|
|NUCLEO_F070RB     |:heavy_check_mark:|:x:|
|LPC1114        |:heavy_check_mark:|:heavy_check_mark:|
|CY8CKIT_064S2_4343W  |:x:|:x:|
|NUCLEO_F072RB     |:heavy_check_mark:|:x:|
|NRF52840_DK      |:heavy_check_mark:|:heavy_check_mark:|
|MTS_DRAGONFLY_F411RE |:heavy_check_mark:|:x:|
|DISCO_H747I_CM4    |:heavy_check_mark:|:x:|
|NUCLEO_F446ZE     |:heavy_check_mark:|:x:|
|HEXIWEAR       |:heavy_check_mark:|:heavy_check_mark:|
|ADV_WISE_1570     |:heavy_check_mark:|:x:|
|NUCLEO_G071RB     |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F446RE     |:heavy_check_mark:|:x:|
|NUCLEO_H743ZI2    |:heavy_check_mark:|:x:|
|NUCLEO_L452RE_P    |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F103RB     |:heavy_check_mark:|:heavy_check_mark:|
|NUMAKER_IOT_M263A   |:heavy_check_mark:|:heavy_check_mark:|
|EFM32GG_STK3700    |:heavy_check_mark:|:x:|
|MTS_MDOT_F411RE    |:heavy_check_mark:|:x:|
|MIMXRT1050_EVK    |:heavy_check_mark:|:heavy_check_mark:|
|DISCO_L072CZ_LRWAN1  |:heavy_check_mark:|:x:|
|WIO_3G        |:heavy_check_mark:|:x:|
|CY8CKIT_062S2_43012  |:x:|:x:|
|EV_COG_AD3029LZ    |:heavy_check_mark:|:heavy_check_mark:|
|NUMAKER_IOT_M487   |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F303K8     |:heavy_check_mark:|:heavy_check_mark:|
|TB_SENSE_12      |:x:|:x:|
|K64F         |:heavy_check_mark:|:heavy_check_mark:|
|MTS_DRAGONFLY_L471QG |:heavy_check_mark:|:x:|
|NUMAKER_PFM_M487   |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_L073RZ     |:heavy_check_mark:|:heavy_check_mark:|
|ARDUINO_NANO33BLE   |:heavy_check_mark:|:heavy_check_mark:|
|NRF52_DK       |:heavy_check_mark:|:x:|
|NUCLEO_F767ZI     |:heavy_check_mark:|:heavy_check_mark:|
|NU_PFM_M2351_NPSA_NS |:heavy_check_mark:|:heavy_check_mark:|
|CYW9P62S1_43438EVB_01 |:x:|:x:|
|ARM_MPS2_M0P     |:heavy_check_mark:|:x:|
|KL46Z         |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_L4R5ZI     |:heavy_check_mark:|:x:|
|LPC54114       |:heavy_check_mark:|:x:|
|DISCO_F429ZI     |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_F412ZG     |:heavy_check_mark:|:x:|
|CY8CPROTO_062S3_4343W |:x:|:x:|
|WIO_EMW3166      |:x:|:x:|
|KL25Z         |:heavy_check_mark:|:heavy_check_mark:|
|NUCLEO_L4R5ZI_P    |:heavy_check_mark:|:x:|
|DISCO_L475VG_IOT01A  |:heavy_check_mark:|:heavy_check_mark:|
|MAX32625MBED     |:heavy_check_mark:|:x:|
|DISCO_H747I_CM7    |:heavy_check_mark:|:x:|
|SDT32625B       |:heavy_check_mark:|:x:|
|LPC1768        |:heavy_check_mark:|:heavy_check_mark:|
|CYSBSYSKIT_01     |:x:|:x:|

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compile 6.13 and GCC 9 2019-q4

#### What version of Mbed-os are you using (tag or sha) ?

6.0.0 - 165be79

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.0

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky-baremetal
mbed compile -t <toolchain> -m <target>
```

evedon, 2020-07-01:
@MarceloSalazar Fix for KW41Z was merged. Can you confirm if this target is now passing (and update the table)?

MarceloSalazar, 2020-07-01:
I'm in the process of updating the table.
I've just tested KW41W with GCC and fails (sha ffeb926a673c)
```
[Fatal Error] NanostackRfPhyKw41z.cpp@17,10: common_functions.h: No such file or directory
[ERROR] ./mbed-os/features/nanostack/targets/TARGET_Freescale/TARGET_KW41Z/NanostackRfPhyKw41z.cpp:17:10: fatal error: common_functions.h: No such file or directory
   17 | #include "common_functions.h"
```

evedon, 2020-07-03:
@LDong-Arm see previous comment from Marcelo

LDong-Arm, 2020-07-03:
The cause is that #13115 (an update from NXP) wasn't rebased onto my fix for bare-metal support (#13119). Now we have duplicated copies of Nanostack drivers for TARGET_KW41Z...
I'll create a PR to fix it.

loverdeg, 2020-08-12:
I need this fixed for CY8CKIT_062_WIFI_BT

From what I gather wifi requires threading but if we can get baremetal at least working for the target that would be a start.

@bentcooke

LDong-Arm, 2020-08-13:
> I need this fixed for CY8CKIT_062_WIFI_BT
> 
> From what I gather wifi requires threading but if we can get baremetal at least working for the target that would be a start.
> 
> @bentcooke

Created a quick fix https://github.com/ARMmbed/mbed-os/pull/13425, feel free to have a try.

evedon, 2020-08-28:
@MarceloSalazar Could you update the table?

MarceloSalazar, 2020-09-21:
I'm in the process of updating the table - please bear with me.
I've re-run the compile tests and see many targets failing, but issues are related to storage.
Can you have a look?

```
$ mbed compile -t GCC_ARM -m DISCO_F413ZH 
[mbed] Working path "/Users/.../Mbed Programs/mbed-os-example-blinky-baremetal" (program)
Building project mbed-os-example-blinky-baremetal (DISCO_F413ZH, GCC_ARM)
Scan: mbed-os-example-blinky-baremetal
Compile [ 98.3%]: qspi_api.c
[Fatal Error] qspi_api.c@28,10: mbed-trace/mbed_trace.h: No such file or directory
```

LDong-Arm, 2020-09-21:
@MarceloSalazar Please see the discussion here: https://github.com/ARMmbed/mbed-os/pull/13617#discussion_r488801025

evedon, 2020-10-22:
@MarceloSalazar mbed-trace is now enabled  by default in bare metal, see #13649 
Can you update the table again?

MarceloSalazar, 2020-11-04:
Table updated

harmut01, 2021-03-08:
@MarceloSalazar with the exception of the targets listed below, nearly all the targets have fixes merged or pending review. Can I update the table once the changes are in? 
|	Target	                |	ARM	                |	GCC_ARM	        |
|	-------	                |  -------		        |   ------		    |
|	ARM_MUSCA_B1	        |	❌	                |	❌	            |
|	 ARM_MUSCA_B1_NS       	|	❌	                |	❌	            |
|	ARM_MUSCA_S1	        |	❌	                |	❌	            |
|	ARM_MUSCA_S1_NS	        |	❌	                |	❌	            |
|	 CY8CKIT_062_BLE       	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CY8CKIT_062_WIFI_BT   	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CY8CKIT_062S2_43012   	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|    CY8CKIT064B0S2_4343W   |	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CY8CPROTO_062_4343W   	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CY8CPROTO_062S3_4343W 	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CYSBSYSKIT_01         	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	CYTFM_064B0S2_4343W	    |	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CYW9P62S1_43012EVB_01 	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 CYW9P62S1_43438EVB_01 	|	❌ *(m/nlib)	    |	❌ *(m/nlib)	|
|	 KW41Z                 	|   ❌	                |	❌	            |
|	 S5JS100               	|	❌	                |	❌	            |
|	 SDT32620B             	|	❌	                |	❌	            |

MarceloSalazar, 2021-03-08:
@harmut01 thanks! I'll re-start the testing once a few PRs get in

harmut01, 2021-03-08:
Awesome! Here are the PRs that have already been merged: https://github.com/ARMmbed/mbed-os/pull/14291, https://github.com/ARMmbed/mbed-os/pull/14300, https://github.com/ARMmbed/mbed-os/pull/14345, https://github.com/ARMmbed/mbed-os/pull/14348, https://github.com/ARMmbed/mbed-os/pull/14349, and https://github.com/ARMmbed/mbed-os/pull/14363. I'll mention this issue in the remaining ones and let you know once they're in.

closed 2021-10-20 by ciarmcom