ARMmbed/mbed-os #13101: WIO_EMW3166 Wi-Fi driver crash on Mbed OS 6

toyowata, 2020-06-12:
### Description of defect

`mbed-os-example-wifi` and `TESTS/network/wifi` test crashed on Mbed OS 6.

[mbed-os-example-wifi]

```
WiFi example
Mbed OS version 6.0.0

Scan:

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: bool(*this)
Location: 0x800DCE5
File: ./mbed-os/platform/Callback.h
```

[TEST-network-wifi]

```
mbedgt: greentea test automation tool ver. 1.7.4
mbedgt: test specification file './BUILD/tests/WIO_EMW3166/GCC_ARM/test_spec.json' (specified with --test-spec option)
mbedgt: using './BUILD/tests/WIO_EMW3166/GCC_ARM/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	| platform_name | platform_name_unique | serial_port              | mount_point      | target_id                                        |
	|---------------|----------------------|--------------------------|------------------|--------------------------------------------------|
	| WIO_EMW3166   | WIO_EMW3166[0]       | /dev/tty.usbmodem1442202 | /Volumes/DAPLINK | 9017000101df135300000000000000000000000097969902 |
mbedgt: processing target 'WIO_EMW3166' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
	| platform_name | platform_name_unique | serial_port                   | mount_point      | target_id                                        |
	|---------------|----------------------|-------------------------------|------------------|--------------------------------------------------|
	| WIO_EMW3166   | WIO_EMW3166[0]       | /dev/tty.usbmodem1442202:9600 | /Volumes/DAPLINK | 9017000101df135300000000000000000000000097969902 |
mbedgt: test case filter (specified with -n option)
	tests-network-wifi
	test filtered in 'tests-network-wifi'
mbedgt: running 1 test for platform 'WIO_EMW3166' and toolchain 'GCC_ARM'
	use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
	found 'host_tests' directory in: 'TESTS/host_tests'
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -m WIO_EMW3166 -p /dev/tty.usbmodem1442202:9600 -f "BUILD/tests/WIO_EMW3166/GCC_ARM/TESTS/network/wifi/wifi.bin" -e "TESTS/host_tests" -d /Volumes/DAPLINK -c default -t 9017000101df135300000000000000000000000097969902 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1591932342.48][HTST][INF] host test executor ver. 0.0.13
[1591932342.48][HTST][INF] copy image onto target...
[1591932342.48][COPY][INF] Waiting up to 60 sec for '9017000101df135300000000000000000000000097969902' mount point (current is '/Volumes/DAPLINK')...
[1591932381.27][HTST][INF] starting host test process...
[1591932381.27][CONN][INF] starting connection process...
[1591932381.27][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1591932381.27][CONN][INF] initializing serial port listener...
[1591932381.27][PLGN][INF] Waiting up to 60 sec for '9017000101df135300000000000000000000000097969902' serial port (current is '/dev/tty.usbmodem1442202')...
[1591932381.27][HTST][INF] setting timeout to: 60 sec
[1591932382.71][SERI][INF] serial(port=/dev/tty.usbmodem1442202, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1591932382.72][SERI][INF] reset device using 'default' plugin...
[1591932383.12][SERI][INF] waiting 1.00 sec after reset
[1591932384.12][SERI][INF] wait for it...
[1591932384.12][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1591932384.12][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1591932384.12][CONN][INF] sending preamble '1040d7ad-ef04-4217-b4b4-8fcd2416d4b5'
[1591932384.12][SERI][TXD] {{__sync;1040d7ad-ef04-4217-b4b4-8fcd2416d4b5}}
[1591932384.25][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1591932384.30][CONN][INF] found SYNC in stream: {{__sync;1040d7ad-ef04-4217-b4b4-8fcd2416d4b5}} it is #0 sent, queued...
[1591932384.30][HTST][INF] sync KV found, uuid=1040d7ad-ef04-4217-b4b4-8fcd2416d4b5, timestamp=1591932384.299365
[1591932384.32][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1591932384.32][HTST][INF] DUT greentea-client version: 1.3.0
[1591932384.34][CONN][INF] found KV pair in stream: {{__timeout;360}}, queued...
[1591932384.34][HTST][INF] setting timeout to: 360 sec
[1591932384.37][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1591932384.37][HTST][INF] host test class: '<class 'mbed_os_tools.test.host_tests.default_auto.DefaultAuto'>'
[... 77 lines cut ...]
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI version 1.10.2

#### How is this defect reproduced ? 

[mbed-os-example-wifi]

```
$ mbed import mbed-os-example-wifi
$ cd mbed-os-example-wifi
# set Wi-Fi credential in mbed_app.json
$ mbed compile -m wio_emw3166 -t gcc_arm -f
```

[TEST-network-wifi]

```
$ git clone -b mbed-os-6.0.0 https://github.com/ARMmbed/mbed-os
$ cd mbed-os
$ mbed test -m wio_emw3166 -t gcc_arm -n TESTS-network-wifi --app-config TESTS/network/wifi/template_mbed_app.txt --compile
$ mbed test -m wio_emw3166 -t gcc_arm -n TESTS-network-wifi --run -v
```

cc @MarceloSalazar @ARMmbed/team-seeed

adbridge, 2020-06-15:
@toyowata this failed to mirror because of 'mbed-os (#165be79392ae, tag: mbed-os-6.0.0)' that is not a single valid tag or sha :). I've fixed it for you.

toyowata, 2020-08-26:
I've got same result on 6.2.1.

```
WiFi example
Mbed OS version 6.2.1

Scan:

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: bool(*this)
Location: 0x800DD85
File: ./mbed-os/platform/include/platform/Callback.h
```