ARMmbed/mbed-os #13102: USBSerial not working using mbed-os6 mbed-lpc1768

vvkaggarwal12, 2020-06-12:
### Description of defect

#### Target(s) affected by this defect ?
 mbed-LPC1768

#### Toolchain(s) (name and version) displaying this defect ?
ARM_GCC

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli

#### How is this defect reproduced ?

adbridge, 2020-06-15:
@vvkaggarwal12 HI you haven't provided any description of the version of mbed-os you were using e.g. mbed-os-6.0.0. You also haven't provided details of how we reproduce the issue . We cannot proceed with this issue until these details are provided. Please revisit the issue header text for an explanation of what is expected for each field.

0xc0170, 2020-06-17:
I'll close the issue as it does not provide sufficient details. Please reopen with more details

closed 2020-06-17 by 0xc0170

vvkaggarwal12, 2020-06-17:
i got the solution as i need to enable DTR signal in terminal for receiving the data over usb in mbed-os 6. Sorry for the late reply