ARMmbed/mbed-os #13104: [custom boards] add default weak TargetBSP_Init function

jeromecoutant, 2020-06-12:
### Description of defect

Discussion started in #13022

Goal is to be able to create a custom board which inherit from another board

Proposition form @kjbracey-arm 

Add in a generic CMSIS startup.S, something like:
````
BL SystemInit
BL UserEarlyInit   // default WEAK empty implementtion
BL __main
````

@AGlass0fMilk 
@MarceloSalazar 
@0xc0170 

#### Target(s) affected by this defect ?

NA
All

#### Toolchain(s) (name and version) displaying this defect ?

NA
All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

NA

jeromecoutant, 2020-06-17:
Solution should take into account #13130 
Solution should be consistent for all targets.

@kjbracey-arm

kjbracey-arm, 2020-06-17:
That "weak" thing wasn't my ideal suggestion - it doesn't quite fit with the general inheritance mechanism. I preferred my "alternate init function" version.

jeromecoutant, 2020-07-02:
> That "weak" thing wasn't my ideal suggestion - it doesn't quite fit with the general inheritance mechanism. I preferred my "alternate init function" version.

Maybe I didn't fully understood, but I preferred the weak  approach as the goal is to add some additional init, not to replace the current SystemInit ?

closed 2021-10-20 by ciarmcom