ARMmbed/mbed-os #13105: EventQueue bug with signed `ms` value in `call_in` and friends

maxgerhardt, 2020-06-12:
### Description of defect

The API for `EventQueue::call_in` takes a `int ms` (`int32_t`) argument which describes when the target callback shall be executed. This is then given to `equeue_event_delay()` which copies the `int ms` value into the `e->target` field, which is a `unsigned` -- a guard against negative millisecond values never happens. Or rather, the type of `ms` should be changed.

See code https://github.com/ARMmbed/mbed-os/blob/master/events/equeue.h#L51

```
    unsigned target;
``` 

and https://github.com/ARMmbed/mbed-os/blob/727cf54873a970017a7be0cf466b0f76bb21caf9/events/source/equeue.c#L577-L582

```cpp
// event functions
void equeue_event_delay(void *p, int ms)
{
    struct equeue_event *e = (struct equeue_event *)p - 1;
    e->target = ms;
}
```

And https://github.com/ARMmbed/mbed-os/blob/master/events/EventQueue.h#L64

```cpp
    using duration = std::chrono::duration<int, std::milli>;
```
notice how you choose the representation `rep` to be `int`, thus also in https://github.com/ARMmbed/mbed-os/blob/master/events/EventQueue.h#L828-L840

```cpp

    template <typename F>
    int call_in(duration ms, F f)
    {
        void *p = equeue_alloc(&_equeue, sizeof(F));
        if (!p) {
            return 0;
        }

        F *e = new (p) F(std::move(f));
        equeue_event_delay(e, ms.count());
        equeue_event_dtor(e, &EventQueue::function_dtor<F>);
        return equeue_post(&_equeue, &EventQueue::function_call<F>, e);
    }
```
the result of `ms.count()` will be `int` ([docs](https://de.cppreference.com/w/cpp/chrono/duration))

And thus a `int` is passed in `equeue_event_delay()` which takes it and writes it into a `unsigned` member -- an internal integer overflow will occur and `-1` would be interpreted as `0xFFFFFFFF` milliseconds aka 49.7 days.

#### Target(s) affected by this defect ?

Every target.

#### Toolchain(s) (name and version) displaying this defect ?

Every toolchain

#### What version of Mbed-os are you using (tag or sha) ?
Latest master branch (727cf54873a970017a7be0cf466b0f76bb21caf9)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
`mbed-cli`

#### How is this defect reproduced ? 

```cpp
#include <mbed.h>
#include <EventQueue.h>
using namespace events;

#define MAX_NUMBER_OF_EVENTS 16
EventQueue ev_queue(MAX_NUMBER_OF_EVENTS * EVENTS_EVENT_SIZE);

void target_function() {
    printf("Target function reached!\n");
}

void main() {
    ev_queue.call_in(-1, &target_function);
    ev_queue.dispatch_forever();
}
```

maxgerhardt, 2020-06-12:
*Updated*

kjbracey-arm, 2020-06-16:
I would say that the prototype of `call_in` should be changed so that is `unsigned`, but that just matches, documents and preserves current behaviour.  The test program would behave exactly the same.

It is possible there are programs out there that are happily calculating delays bigger than 24 days, feeding them to `call_in` via a wrapped `int`, and relying on this behaviour. I don't think I'd want to halve the range here.

Pretty much all other core 32-bit relative-time APIs do take `uint32_t` or a `duration<uint32_t, milli>` - these signed millisecond ones are anomalies.

maxgerhardt, 2020-06-16:
>I would say that the prototype of call_in should be changed so that is unsigned, but that just matches, documents and preserves current behaviour. The test program would behave exactly the same.

Indeed I think that is what should be done here -- just match the signature of the outer functions to what is going in on the inside. The `equeue_event_delay()` function shouldn't take an `int ms` and write it into `unsigned target`, but a `unsigned int` instead. 

But is that really documented? Looking into https://os.mbed.com/docs/mbed-os/v6.0/mbed-os-api-doxy/classevents_1_1_event_queue.html, there's no mention of a 49 day limit (or that negative values will be interpreted as being beyond 24 days).

kjbracey-arm, 2020-06-16:
No, it's not documented. But changing the prototype to `unsigned` would serve as documentation. "Pass -1 to this, and you'll get 2^32-1 ms".

0xc0170, 2020-06-16:
cc @ARMmbed/mbed-os-core

evedon, 2020-06-17:
I also agree with changing the prototype to `unsigned`. The fix could go in a patch release as this would not affect current behaviour.

maxgerhardt, 2020-06-17:
Your CI bot is really broken if it can't recognize the text `727cf54873a970017a7be0cf466b0f76bb21caf9` as commit hash. Trying with a `#` prefix now.

0xc0170, 2020-06-17:
@maxgerhardt Yes, there is an issue. its rather related to released version and mapping sha to it. rather than not recognizing the sha itself. will be fixed

adbridge, 2020-06-17:
@maxgerhardt as @0xc0170 said it is because the bot is designed to determine the last released version of software to contain the sha and as this sha is so far in unreleased code, it can't obviously find a release. This is a known issue that we are working on, please bear with us.

maxgerhardt, 2020-06-17:
If it isn't critical then it's not in issue. The bot did however say that without a correctly formed SHA, it wouldn't mirror it to the internal repo where the actual work is done. Ignoring the bot then.

maxgerhardt, 2020-06-17:
Also, closing upon merging of patch into the next path release.

adbridge, 2021-01-19:
Looking at this further it definitely looks like it should be the equeue_event_delay() function that should be changed to take an unsigned rather than an int (rather than the call_in() function which was mentioned above at one point. This is because equeue_event_delay() is also used in other places internally (such as UserAllocatedEvent methods so they would exhibit the same behaviour as call_in(). I'll put up a PR to do this.

kjbracey-arm, 2021-01-27:
Okay, Anna's been looking at this a lot, and I've been reviewing it. I believe the original report is invalid - this statement is not true:

> -1 would be interpreted as 0xFFFFFFFF milliseconds aka 49.7 days.

Anna's checked and this does not happen. In fact, the `-1` is interpreted as `0xFFFFFFFF (mod 2^32)`. That's then added to "now" giving the result `now + 0xFFFFFFFF (mod 2^32)` = `now - 1 (mod 2^32)`. The time comparisons then decide that is in the past causing an immediate callback.

There's a little confusion because that `e->target` field is primarily used to store absolute (modulo 2^32) timestamps. `equeue_post` converts that field from relative to absolute time.

        e->target = tick + e->target;

So `int` is actually the correct type for `call_in` - it will work correctly for most of the signed range, including immediate callback for negative, with potential overflow issues only occurring near `INT_MAX` or `INT_MIN`.

As such I withdraw my suggestion of changing to `unsigned` - that was on the assumption that it was possible to request callbacks more than `INT_MAX` in the future, as the original report suggested.

I think there's nothing to do here apart from possibly documentation.

evedon, 2021-01-29:
Thanks for looking into this further Kevin. 
It's good to know that the code is mostly behaving correctly and for now, we should add a documentation note as you suggest.

But it would be good to cleanup this API in a future release; `equeue_call_in` `equeue_call_every` `equeue_event_delay` and `equeue_event_period` should all take an `unsigned`.