ARMmbed/mbed-os #13106: LPC1114 doesn't compile with normal blinky

MarceloSalazar, 2020-06-12:
### Description of defect

LPC1114 fails to compile [blinky](github.com/ARMmbed/mbed-os-example-blinky) application with latest version of Mbed OS 6 (master)

 - GCC Error
   `BUILD/LPC1114/GCC_ARM/.link_script.ld:77 cannot move location counter backwards (from 0000000010001c38 to 0000000010000c00)`
 - ARM
   `Error: L6406E: No space in execution regions with .ANY selector matching BUILD/LPC1114/ARM/mbed-os`

Is this expected to build?
If not, the tools should inform that this isn't supported or is a bare-metal only target

#### Target(s) affected by this defect ?

LPC1114

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6.13 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

master (727cf54873a9)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m LPC1114
```

@ARMmbed/team-seeed @toyowata

evedon, 2020-06-12:
This target does not have enough memory for the rtos profile. 
The `supported_application_profiles` configuration is only used for documentation purposes currently. This will be addressed by the new tools.

MarceloSalazar, 2020-06-17:
I've realized this is not a target issue as it's not expected to work due to limited memory.
However, the tools should display a message indicating that the target doesn't support the full profile.

evedon, 2020-06-19:
> I've realized this is not a target issue as it's not expected to work due to limited memory.
> However, the tools should display a message indicating that the target doesn't support the full profile.

This functionality should come in the new tools. 
In the meantime, we could add some information in blinky's README to indicate which targets are supported. Targets which don't list `full` in `supported_application_profiles configuration` are not supported.

LDong-Arm, 2020-09-17:
PR created: https://github.com/ARMmbed/mbed-os-example-blinky/pull/238

LDong-Arm, 2020-09-18:
@MarceloSalazar The README update has been merged but this issue isn't automatically closed - presumably because it's a different project. Are you happy to close it?

MarceloSalazar, 2020-09-18:
That's fine. Thanks!

closed 2020-09-18 by MarceloSalazar