ARMmbed/mbed-os #13107: NUCLEO_L432KC doesn't compile on normal blinky with ARM compiler

MarceloSalazar, 2020-06-12:
### Description of defect

NUCLEO_L432KC fails to compile [blinky](github.com/ARMmbed/mbed-os-example-blinky) application with latest version of Mbed OS 6 (master)

 - GCC OK
 - ARM FAIL
    ```
    [Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base
    [Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Limit
    [Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length 
   ```

Is this expected to build?
If not, the tools should inform that this isn't supported or is a bare-metal only target

#### Target(s) affected by this defect ?

NUCLEO_L432KC

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6.13 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?
<!--
    For a released version please provide the release tag (this can be found as per the instructions below)

    mbed-os version can be found in /platform/mbed_version.h. The tag can be reconstructed as follows:
    mbed-os-MBED_MAJOR_VERSION.MBED_MINOR_VERSION.MBED_PATCH_VERSION
 
    Master branch is indicated by 'mbed-os-99.99.99
    
    For an issue found on Master please provide the sha being used.
-->

master (727cf54873a9)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
mbed compile -t ARM -m NUCLEO_L432KC
```

@ARMmbed/team-st-mcd

jeromecoutant, 2020-06-12:
Waiting for #13001

jeromecoutant, 2020-06-12:
@MarceloSalazar Could you check again now?

MarceloSalazar, 2020-06-12:
@jeromecoutant Thanks for the PR!
I confirm the NUCLEO_L432KC targets compiles OK on both blinky and blinky-baremetal using both ARM and GCC_ARM :smile:

closed 2020-06-12 by MarceloSalazar