ARMmbed/mbed-os #13108: Cypress targets fail to compile blinky-baremetal in Mbed OS 6

MarceloSalazar, 2020-06-12:
### Description of defect

Cypress targets fail to compile blinky-baremetal in Mbed OS 6

```
[Fatal Error] cyhal_system.c@30,10: cyabs_rtos.h: No such file or directory
[ERROR] ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/psoc6csp/hal/source/cyhal_system.c:30:10: fatal error: cyabs_rtos.h: No such file or directory
   30 | #include "cyabs_rtos.h"

[Fatal Error] scl_buffer_api.h@28,10: 'memp.h' file not found
[ERROR] In file included from ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/COMPONENT_SCL/src/scl_buffer_api.c:18:
./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/COMPONENT_SCL/src/include/scl_buffer_api.h:28:10: fatal error: 'memp.h' file not found
```

#### Target(s) affected by this defect ?

- CY8CKIT_062_BLE
- CY8CPROTO_062_4343W
- CY8CPROTO_062S3_4343W
- CYW9P62S1_43012EVB_01
- CY8CKIT_062_WIFI_BT
- CY8CPROTO_064_SB
- CY8CKIT_064S2_4343W
- CY8CKIT_062S2_43012
- CYW9P62S1_43438EVB_01
- CYSBSYSKIT_01

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os master (d0d35c274362)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@ARMmbed/team-cypress @maclobdell

LDong-Arm, 2020-08-13:
@MarceloSalazar 
The fix is a quick and easy one: https://github.com/ARMmbed/mbed-os/pull/13425

MarceloSalazar, 2020-11-04:
Issue remains for CY8CKIT_064S2_4343W

LDong-Arm, 2020-11-04:
> * CYTFM_064B0S2_4343W

@MarceloSalazar I totally missed one of the issues in the PR description!

PR created: #13864. I don't see CY8CKIT_064S2_4343W in `targets.json` so only the other two are fixed. How to enable that target?

MarceloSalazar, 2020-11-04:
Sorry there was a typo, it's this one CY8CKIT064B0S2_4343W

LDong-Arm, 2020-11-05:
> ```
> Elf2Bin: mbed-os-example-blinky-baremetal
> Post-Build: mbed-os-example-blinky-baremetal
> M0 core image file found: ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CKIT064B0S2_4343W/device/COMPONENT_CM4/hex/psoc6_02_cm0p_secure.hex.
> Policy file /Users/.../Mbed Programs/mbed-os-example-blinky-baremetal/policy_single_CM0_CM4.json not found. Aborting.
> [ERROR] Required policy file not found.
> [mbed] ERROR: "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/bin/python" returned error.
>        Code: 1
>        Path: "/Users/.../Mbed Programs/mbed-os-example-blinky-baremetal"
>        Command: "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/bin/python -u /Users/.../Mbed Programs/mbed-os-example-blinky-baremetal/mbed-os/tools/make.py -t GCC_ARM -m CY8CKIT064B0S2_4343W --source . --build ./BUILD/CY8CKIT064B0S2_4343W/GCC_ARM -c"
>        Tip: You could retry the last command with "-v" flag for verbose output
> ```

To build any application for this Cypress target, you need to manually generate policy files as https://github.com/ARMmbed/mbed-os/pull/13122#issuecomment-647731540 There are a few python dependency to install too.

PR #13864 (2020-11-11): Fix bare metal builds on Cypress targets

### Summary of changes 

Fixes: #13108

This PR fixes bare metal build of two Cypress targets:
* CYSBSYSKIT_01: `COMPONENT_SCL` depends on the EMAC/LWIP stack which is absent on bare metal. Add an `mbed_lib.json` to disable its build in the bare metal profile.
* CYTFM_064B0S2_4343W: Make `cytfm_flash*` dependent on TF-M and PSA presence.

#### Impact of changes
closed 2020-11-11 by 0xc0170