ARMmbed/mbed-os #13111: S5JS100 target doesn't compile with blinky-baremetal

MarceloSalazar, 2020-06-12:
### Description of defect

S5JS100 target fails to compile blinky-baremetal in Mbed OS 6

GCC Error

```
[Fatal Error] modem_io_device.h@24,10: pbuf.h: No such file or directory
[ERROR] In file included from ./mbed-os/targets/TARGET_Samsung/TARGET_SIDK_S5JS100/dcxo_update.cpp:20:
./mbed-os/targets/TARGET_Samsung/TARGET_SIDK_S5JS100/modem/modem_io_device.h:24:10: fatal error: pbuf.h: No such file or directory
   24 | #include "pbuf.h"
```

Arm error
```
[Fatal Error] s5js100_systemreset.c@51,10: 'rtx_os.h' file not found
[ERROR] ./mbed-os/targets/TARGET_Samsung/TARGET_SIDK_S5JS100/device/s5js100_systemreset.c:51:10: fatal error: 'rtx_os.h' file not found
#include "rtx_os.h"
```

#### Target(s) affected by this defect ?

S5JS100

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os master (d0d35c274362)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@armmbed/team-samsung

0xc0170, 2020-06-16:
`rtx_os.h` should not be included by a target,  that should rely on rtos header rather . A question is why `s5js100_systemreset` implementation needs it..

LDong-Arm, 2020-06-16:
> `rtx_os.h` should not be included by a target, that should rely on rtos header rather . A question is why `s5js100_systemreset` implementation needs it..

Apparently it doesn't need it. Same for `pbuf.h`. They might be leftover code I guess.

After removing those redundant includes, we hit other issues in `TARGET_SIDK_S5JS100`, mainly
* Source files include `mbed_trace.h` even when tracing is not enabled (which is the default).
* Vendor-specific features (e.g. `ModemIoDevice`) don't check for `MBED_CONF_RTOS_PRESENT`. We can safely add the checks as they are not used by Mbed OS.

LDong-Arm, 2020-06-17:
Okay, this target's `hal_sleep()` and `hal_deepsleep()` implementations _require_ RTOS. The [power management](https://github.com/ARMmbed/mbed-os/blob/b1629b7e5962f9fd0382e6b9bb83e97b6f9f6b13/targets/TARGET_Samsung/TARGET_SIDK_S5JS100/s5js100_dcxo.cpp) uses APIs like `Thread` and `Semaphore`.

@MarceloSalazar @0xc0170 Is bare metal support a requirement for all targets?

LDong-Arm, 2020-06-17:
I raised a question in the original PR that added the target https://github.com/ARMmbed/mbed-os/pull/12106/files#r441556994

MarceloSalazar, 2020-06-17:
@LDong-Arm thanks for looking into this

> Is bare metal support a requirement for all targets?

Not at this time. We'd like to understand the reason behind the error and check with @ARMmbed/team-samsung the possibility to fix this it at some point soon.

In the meantime, we should consider adding an enhancement to the tools to inform that a target doesn't support baremetal or the full profile (FYI @rwalton-arm )

LDong-Arm, 2020-07-10:
#13265 created

0xc0170, 2021-05-12:
Fixed via #14372

closed 2021-05-12 by 0xc0170