ARMmbed/mbed-os #13116: Compiling a project with mbed-os 6.0.0 gives an error

jadhavshital, 2020-06-15:
Compiling a project with mbed-os 6.0.0 gives expected value in expression error.
Kindly refer below screenshot for more details.
[image]

Please let me know the solution for this.

jadhavshital, 2020-06-15:
Hi @ciarmcom, I have noticed that in the previous version of mbed-os the line was `#ifdef MBED_MEM_TRACING_ENABLED` in `mbed_alloc_wrappers.cpp` but it is now changed to `#if MBED_MEM_TRACING_ENABLED`.
I guess this is the bug in new mbed-os v6.0.0
Please correct me if I am missing something.

0xc0170, 2020-06-16:
@jadhavshital Please use issue template for better description of the bug.

What application are you building? Do you use tracing - is it enabled or set anywhere? Previously it was enought to define it to enable, and remove the config to disable. Since 6.0, you should not only define it but also set a value to enable/disable it.

adbridge, 2020-11-17:
@jadhavshital could you please provide the information requested by @0xc0170 or close this issue if you no longer see the problem. Thanks.

LDong-Arm, 2021-05-26:
I think it could be to do with the compiler version. At least for relatively recent compilers, `#if MBED_MEM_TRACING_ENABLED` automatically checks if the macro is defined and evaluates to false if it's undefined.

The recommended versions of toolchains are listed here: https://os.mbed.com/docs/mbed-os/v6.10/build-tools/index.html