ARMmbed/mbed-os #13121: UREGENT: Multitech MDot: Flashing MDot firmware compiled using MBED Online compiler causes CRC Errors

yogeshk19, 2020-06-15:
### Description of defect
We are compiling a custom application using the MBED online compiler and we have always been able to flash the firmware without any issues however as of today and we are getting CRC errors as shown below, when flashing the MDot module using tera term and ymodem interface. I am using the MDot library 3.1.0 and mbed-os-5.7.7 to compile my custom firmware. We are blocked on generating production builds and would like to receive some assistance ASAP.

#### Target(s) affected by this defect ?
MTS_MDOT_F411RE

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?
[16bac10](https://github.com/ARMmbed/mbed-os/commit/16bac101a6b7b4724023dcf86ece1548e3a23cbf)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
MBED ONLINE COMPILER

#### How is this defect reproduced ? 
Compile any application targeted for the above target using Multitech MDot library 3.1.0 and mbed-os-5.7.7 and once the application is successfully compiled, use the downloaded **<application name>. bin** file to flash the MDot and you should see the follow error mentioned below.

> bootloader :> upgrade ymodem
> [INFO] Starting upgrade
> [INFO] Starting YMODEM file transfer
> [INFO] YMODEM file transfer succeeded
> [INFO] **appended crc: 50050000**
> [INFO] **calculated crc: 2793190355**
> [ERROR] File validation failed

Thanks,
Yogesh

reissjason, 2020-06-15:
CRC erros may be caused by bootloader issues when compiling with mbed-os-5 targets.

yogeshk19, 2020-06-15:
I removed the urgent from the title of the issue as I have managed to get this working using the offline compiler, however it would be great if we can have this issue resolved in the online compiler as well.

Thanks,
Yogesh

0xc0170, 2020-06-16:
> I removed the urgent from the title of the issue as I have managed to get this working using the offline compiler, however it would be great if we can have this issue resolved in the online compiler as well.

No ideas what is the difference here? I assume changing the tools in the online compiler (means  that the recent changes on master could be one to blame?

cc @ARMmbed/mbed-os-tools

rwalton-arm, 2020-06-16:
> No ideas what is the difference here? I assume changing the tools in the online compiler (means that the recent changes on master could be one to blame?
>
> cc @ARMmbed/mbed-os-tools

The difference is the offline compiler uses the Mbed 5.7 tools as they're bundled with Mbed OS, meaning if you have version 5.7 of Mbed OS checked out it will use the Mbed 5.7 tools. The online compiler uses v6.0.0 tools with some patches to try and support older Mbed versions due to problems in the online compiler architecture (it expects to use one set of tools to build every version of mbed-os). The recent changes on master were to revert a commit that removed the post-build hooks for MTS_MDOT, as the post build step was failing online. I think the problem is we need to revert yet more changes in the online compiler branch of the tools to avoid this CRC error.

theckmts, 2020-06-16:
The difference is probably due to the old MDot bootloader being removed from mbed-os-6 tools.  

Recently the MDot post-build script (MTSCode.combine_bins_mts_dot) was restored so projects with mbed-os-5 can be compiled online; however, the bootloader binary was not restored.  So, while the compilation is successful, the binary will not boot and since the bootloader binary isn't found a CRC is not appended to the binary.

0xc0170, 2020-06-17:
@theckmts can you send PR to address this on master (to enable tools to be usable in the online compiler) ?

theckmts, 2020-06-17:
This pull-request is for master but the post-build script restore change is on the tools-release-6.0.0 branch.

rwalton-arm, 2020-06-17:
Thanks for the PR. The post build script revert is also on master, but we will need to create another tools-release tag for the online build service after this is fixed. I noticed some other bootloader binaries were removed on the run up to mbed 6 so I'll also need to restore those before creating a new tools-release.

closed 2020-06-18 by 0xc0170

adbridge, 2020-07-03:
> Thanks for the PR. The post build script revert is also on master, but we will need to create another tools-release tag for the online build service after this is fixed. I noticed some other [...]

@rwalton-arm Has this additional work been done ? There is a comment on the forum from 7 days ago that this is still not working in the online compiler ?

rwalton-arm, 2020-07-06:
> @rwalton-arm Has this additional work been done ? There is a comment on the forum from 7 days ago that this is still not working in the online compiler ?

@adbridge This particular issue should've been fixed in the 6.1 release of the online tools. They were deployed to the online compiler 6 days ago.

theckmts, 2020-07-06:
> @adbridge This particular issue should've been fixed in the 6.1 release of the online tools. They were deployed to the online compiler 6 days ago.

I just compiled a blinky application targeting mbed-os 5.15.4 with the online compiler and verified the bootloader is merged and it runs on an MDot.

PR #13135 (2020-06-18): Restore MTS_MDOT_F411RE bootloader binary
### Summary of changes 
Restore MTS_MDOT_F411RE bootloader binary in tools to fix problems encountered with online compiler and mbed-os 5.x projects.

Fixes #13121

#### Impact of changes 

#### Migration actions required 

### Documentation 
None

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [X] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (