ARMmbed/mbed-os #13123: PDMC Client Stuck at UPD2_STATE_3 during FOTA

lauleehong, 2020-06-16:
### Description of defect
- ARMC6 compiler optimization causes defective behavior when doing FOTA via PDMC 4.4.0 using mbed-os-example-pelion using standard release build.

#### Target(s) affected by this defect ?
- Custom STM32F767ZI

#### Toolchain(s) (name and version) displaying this defect ?
- ARMC 6.13

#### What version of Mbed-os are you using (tag or sha) ?
- mbedos 6.0.0 (165be79)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
- Mbed Studio 1.0 on MacOS

#### How is this defect reproduced ? 
- Connect using STM32F767ZI with BG96 or ESP32
- Compile mbed-os-example-pelion (PDMC 4.4.0) using release build.
- Launch a firmware-over-the-air update from Pelion portal, and the client would be stuck at Awaiting download approval (UPD2_STATE_3)

`[INFO][mClt]: M2MNsdlInterface::observation_to_be_sent() - send already in progress
[INFO][CELL]: Socket 0 recv 5 bytes
[INFO][CELL]: Socket 0 recv 20 bytes
[INFO][COAP]: ======== Incoming CoAP package ========
[INFO][COAP]: Status:           COAP_STATUS_OK
[INFO][COAP]: Code:             COAP_MSG_CODE_EMPTY
[INFO][COAP]: Type:             COAP_MSG_TYPE_ACKNOWLEDGEMENT
[INFO][COAP]: Id:               22268
[INFO][COAP]: Payload len:      0
[INFO][COAP]: ======== End of CoAP package ========
[INFO][mClt]: M2MNsdlInterface::send_resource_observation - uri 10252/0/2
[INFO][COAP]: ======== Outgoing CoAP package ========
[INFO][COAP]: Status:           COAP_STATUS_OK
[INFO][COAP]: Code:             COAP_MSG_CODE_RESPONSE_CONTENT
[INFO][COAP]: Type:             COAP_MSG_TYPE_CONFIRMABLE
[INFO][COAP]: Id:               22269
[INFO][COAP]: Token:            0:2
[INFO][COAP]: Content-type:     0
[INFO][COAP]: Payload len:      1
[INFO][COAP]: Max-age:          0
[INFO][COAP]: Observe:          8
[INFO][COAP]: ======== End of CoAP package ========
[INFO][CELL]: Socket 0 sent 33 bytes to 52.192.171.84 port 5684
[INFO][PELION_CLIENT]: Firmware download requested
[INFO][PELION_CLIENT]: Authorization granted
[INFO][PELION_CLIENT]: Priority of request 0
[INFO][mClt]: M2MNsdlInterface::observation_to_be_sent() - send already in progress

`

@soleilplanet @desmond-blue

amq, 2020-06-19:
Is everything working as expected with Mbed OS 5.15?

lauleehong, 2020-06-22:
Same issue with 5.15.4

0xc0170, 2020-06-23:
> STM32F767ZI

This looks like a generic issue but we should be certain. Is only one target affected or more? If it's just one family, we would know this is in targets code.

cc @ARMmbed/team-st-mcd (if you are aware of any similar issue previously?)

lauleehong, 2020-06-24:
I've only tested on my custom F767ZI board (identical to NUCLEO F767ZI, mostly header re-arrangements). However, @soleilplanet did manage to run it on his NUCELO F767ZI board.

0xc0170, 2020-06-25:
@soleilplanet would you be able to test this with complete different family, even vendor change would be better. 

Without further details, this is very generic issue that needs more investigation to find out the root case - at least the component that is causing it (target/pelion/core or similar).

teetak01, 2020-12-10:
@lauleehong have you tried with latest master of Mbed OS? This looks similar to https://github.com/ARMmbed/mbed-os/issues/13772 and https://github.com/ARMmbed/mbed-os/issues/13792

lauleehong, 2020-12-14:
@teetak01 Yes! I've just tested pdmc 4.7.0 using mbed-os-example-pelion 1.3.0, with mbed-os 6.5.0 - Issue no longer persist.

closed 2020-12-14 by lauleehong