ARMmbed/mbed-os #13127: [STM32WB] LSE clock initializing error

felipeLeast, 2020-06-16:
### Description of defect
Hi, we are using a custom PCB board with stm32wb55 recomended hardware design and uploading mbed into it. We've never had a problem so far, on our latest manufacturing we had to change the LSE external crystal because of market outage. We've picked a very similar (actually even more precise) crystal and now gives us an error every time system_clock initializes.

```
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x800673D
Error Value: 0x0
Current Thread: <unnamed> Id: 0x0 Entry: 0x8002F2D StackSize: 0x0 StackMem: 0x8006861 SP: 0x2002FEA4 
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_WB55RG
-- MbedOS Error Info --
HAL_RCC_OscConfig error
```

We've follow the error down and it's in the HAL driver for STM32WBxx that timeout on initialize. The thing is that if we don't use Mbed and compile a raw application using STM32Cube the program runs as it is supposed.

We've narrowed the problem to the crystal by resoldering an old one over and works normally. The only difference from Mbed and Cube raw application is the file `hal_tick_overrides.c` which can't be deactivate because produces a core error on the RTOS.

Follows the link for both crystals:
* [New crystal](https://br.mouser.com/ProductDetail/ABRACON/ABS07AIG-32768kHz-9-1-T?qs=%2Fha2pyFaduhY6YKT5CDtkJhB03ywq2BVs7B%2FBt4Ee%2FtFUg3CNxOdIA07c4zgoWHS)
* [Old crystal](https://br.mouser.com/ProductDetail/Epson-Timing/FC-135-327680KA-AC?qs=HV%2FSjD3ivR%2FPPRvbL4Usqg%3D%3D)

Also works if we change the system_clock and lp_ticker to use LSI1 or LSI2 clock sources. We believe that the problem is in the HAL_GetTick override function, but we can't also comment that because throws another problem on top/

#### Target(s) affected by this defect ?
NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
Change the crystal from nucleo dev board

@jeromecoutant @LMESTM

LMESTM, 2020-06-17:
@felipeLeast  the thrown error in your log is "HAL_RCC_OscConfig error" which comes from file 
targets/TARGET_STM/TARGET_STM32WB/TARGET_STM32WB55xx/TARGET_NUCLEO_WB55RG/system_clock.c

There are several types of examples in Cube, can you point to the SystemClock_Config function that you have been using. Is it for instance like https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/P-NUCLEO-WB55.Nucleo/Examples/ADC/ADC_AnalogWatchdog/Src/main.c#L213 
or 
https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_Beacon/Core/Src/main.c#L181
?

In the last one I can see a specific call to 
```
  __HAL_RCC_LSEDRIVE_CONFIG(RCC_LSEDRIVE_LOW);
```
Maybe that is worth trying to add it. 

I don't have any LSE expertise, but if you have a working clock configuration from the Cube examples, you should be able to use the same in the SetSysClock function for your custom board.

jeromecoutant, 2020-07-02:
Hi

For more details, there is the AN2867 Application note from ST:
https://www.st.com/resource/en/application_note/cd00221665-oscillator-design-guide-for-stm8afals-stm32-mcus-and-mpus-stmicroelectronics.pdf

jeromecoutant, 2020-08-12:
Hi maybe we could close this issue ?
Thx

0xc0170, 2020-08-12:
@felipeLeast if you resolved an issue or any update, let us know. Otherwise we would close this.

felipeLeast, 2020-08-12:
Hi @jeromecoutant and @0xc0170 

I actually couldn't resolve this issue yet, maybe there is some relation to my board, it's one I soldered myself, but even though I may had some soldering error the program from CubeIDE still worked while the one from Mbed didn't, I've went deep searching the problem even to mbed initializations, tried replacing everything to cube, I even tried changing the .s file to start in another method so I could initialize exactly like CubeIDE, and still didn't work. The only two differences I found between CubeIDE and Mbed are the .ld file which has some incongruences with the one generated by cube, like some of the aligns. And the hal_tick_overrides.c which doesn't exist in cubeIDE and I can't remove or comment on mbed because crashes.

This was the last I debugged, I don't know if it's just a hardware issue but I believe is a mix between both. There were something wrong in the hardware but the mbed code haven't found a way to fix it. My suspicion is that CubeIDE found the LSE problem and start using internal clock instead, whilest mbed couldn't recover from LSE initialization problem.

Thank you,

jeromecoutant, 2020-08-13:
But how could we help ?

felipeLeast, 2020-08-13:
I still didn't had the time to do this test, but maybe the Mbed code won't work without LSE crystal, and I believe it should, I don't know if this test were made. What strike me as odd was the fact that in cubeIDE I could communicate with the board, make leds blink, use serial and SPI and on Mbed I didn't, so must have some difference between both. It may cause some headache in the future.

LMESTM, 2020-08-13:
> I still didn't had the time to do this test, but maybe the Mbed code won't work without LSE crystal, and I believe it should, I don't know if this test were made. What strike me as odd was the fact [...]

Indeed CubeIDE peripheral examples don't use LSE by default.
But any BLE example from cubeIDE makes use of LSE and I believe LSE is required for BLE.
So in mbed we decided to always use LSE. Do you plan to use STM32WB without the BLE feature ?

0xc0170, 2020-09-14:
I'll close this as resolved. We need more details for this to be reopened as indicated by @jeromecoutant and @LMESTM , thanks!

closed 2020-09-14 by 0xc0170