ARMmbed/mbed-os #13129: Error with mbed 6.0, gcc 9.2 and GD25Q16 QSPI Flash -- incorrect SFDP parsing from QSPIFBlock

keithmwheeler, 2020-06-16:
### Description of defect

#### Target(s) affected by this defect ?
Custom CYW9P62S1_43438EVB_01 with GD25Q16 flash.

#### Toolchain(s) (name and version) displaying this defect ?
GCC/ARM 9.2

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
Attempt to initialize this device, block device init returns -4002. 

Using a debugger and stepping in to the code, it appears that in QSPIFBlockDevice.cpp the parsing of SFDP tables is not handled correctly for older devices. Specifically the _sfdp_detect_reset_protocal_and_reset method is looking at uninitialized memory and does not get the correct reset type for the 25Q16. This is a bug and mbed should at the least report an error if the SFDP version is old; this could be handled by a length check of the received table, as is done elsewhere for other QSPI elements. Without an error reporting the legacy SFDP table type, a user attempting to use this flash must go through a lengthy debug effort to understand the failure root cause.
 
In addition, since the SFDP parser is not written to handle this type of table, the driver does not correctly handle the vendor quirks of the 25Q16.

0xc0170, 2020-06-17:
cc @ARMmbed/team-cypress

loverdeg-ep, 2020-06-18:
@maclobdell this is an issue that could use some prioritization.

loverdeg-ep, 2020-06-24:
@0xc0170 I don't have the evidence in front of me but we believe this to be an issue with the block device driver from an Mbed standpoint.

loverdeg-ep, 2020-06-24:
@ARMmbed/mbed-os-storage

LDong-Arm, 2020-08-07:
@keithmwheeler Many thanks for raising this, your analysis makes complete sense to me.

GD25Q16 does have a legacy SDFP parameter table (1.0). In fact, their [data sheets](http://www.elm-tech.com/en/products/spi-flash-memory/gd25q16/gd25q16.pdf) lists a _vendor-specific_ parameter table (Table 5) outside SDFP standard, and it says "Reset Enable(66H) before Reset cmd"

I've created a pull request: #13399. If you would like a try, please add
```json
    "target_overrides": {
        "CYW9P62S1_43438EVB_01": {
            "enable-and-reset": true
        }
    }
```
to your `mbed_app.json`.

Thanks.

@loverdeg, @ladislas: FYI