ARMmbed/mbed-os #13136: Gigadevice targets don't compile with blinky-baremetal

MarceloSalazar, 2020-06-17:
### Description of defect

Gigadevice targets fail to compile blinky-baremetal in Mbed OS 6

```
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/GD32_F307VG/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/GD32_F307VG/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/GD32_F307VG/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/GD32_F307VG/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/GD32_F307VG/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/GD32_F307VG/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
```

#### Target(s) affected by this defect ?

- GD32_F307VG
- GD32_F450ZI

GCC and ARM Errors

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

talorion, 2020-08-26:
can this be closed?
works on my machine
mbed os: afcf91f3314d82ce8d555f4a891c9fd8db13910f
GCC 9
[GD32_F307VG.txt](https://github.com/ARMmbed/mbed-os/files/5132414/GD32_F307VG.txt)
[GD32_F450ZI.txt](https://github.com/ARMmbed/mbed-os/files/5132415/GD32_F450ZI.txt)

evedon, 2020-10-27:
> can this be closed?
> works on my machine
> mbed os: [afcf91f](https://github.com/ARMmbed/mbed-os/commit/afcf91f3314d82ce8d555f4a891c9fd8db13910f)
> GCC 9
> [GD32_F307VG.txt](https://github.com/ARMmbed/mbed-os/files/5132414/GD32_F307VG.txt)
> [GD32_F450ZI.txt](https://github.com/ARMmbed/mbed-os/files/5132415/GD32_F450ZI.txt)

Indeed, it works with mbed-os (#8ef0a435b235, tags: mbed-os-6.4.0, mbed-os-6.4.0-rc1)
Closing this issue.

closed 2020-10-27 by evedon