ARMmbed/mbed-os #13138: Arm FVP targets don't compile with blinky-baremetal

MarceloSalazar, 2020-06-17:
### Description of defect

Few Silabs targets fail to compile blinky-baremetal in Mbed OS 6

```
Building project mbed-os-example-blinky-baremetal (ARM_MPS2_M4, ARMC6)
Scan: mbed-os-example-blinky-baremetal
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
Link: mbed-os-example-blinky-baremetal
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/ARM_MPS2_M4/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/ARM_MPS2_M4/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/ARM_MPS2_M4/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/ARM_MPS2_M4/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/ARM_MPS2_M4/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/ARM_MPS2_M4/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
```

#### Target(s) affected by this defect ?

- FVP_MPS2_M0
- FVP_MPS2_M0P
- FVP_MPS2_M3
- FVP_MPS2_M4
- FVP_MPS2_M7

GCC and ARM Errors

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@jamesbeyond

talorion, 2020-08-26:
can this be closed?
works on my machine
mbed os: afcf91f3314d82ce8d555f4a891c9fd8db13910f
[FVP_MPS2_M0.txt](https://github.com/ARMmbed/mbed-os/files/5132354/FVP_MPS2_M0.txt)
[FVP_MPS2_M0P.txt](https://github.com/ARMmbed/mbed-os/files/5132355/FVP_MPS2_M0P.txt)
[FVP_MPS2_M3.txt](https://github.com/ARMmbed/mbed-os/files/5132356/FVP_MPS2_M3.txt)
[FVP_MPS2_M4.txt](https://github.com/ARMmbed/mbed-os/files/5132357/FVP_MPS2_M4.txt)
[FVP_MPS2_M7.txt](https://github.com/ARMmbed/mbed-os/files/5132358/FVP_MPS2_M7.txt)

MarceloSalazar, 2020-11-04:
Tested again on 29th October and issue remains
See report  https://github.com/ARMmbed/mbed-os/issues/13099

evedon, 2020-11-04:
The Arm scatter file needs to be updated with a 2-region memory model for heap and stack.

jamesbeyond, 2020-11-04:
yep, I looked into fixing this defect some time ago. but I would not finished this.
I am able to update the scatter file into 2-region memory model, able to compile it. however the image would not run with Fastmodel `-f` loading. but if I load the image with debugger, e.g. uVision or ARM-DS, the same image works just fine.

Haven't been able to dig it deeper