ARMmbed/mbed-os #13139: Maxim targets don't compile with blinky-baremetal

MarceloSalazar, 2020-06-17:
### Description of defect

Few Silabs targets fail to compile blinky-baremetal in Mbed OS 6

```
Link: mbed-os-example-blinky-baremetal
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/MAX32600MBED/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/MAX32600MBED/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/MAX32600MBED/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/MAX32600MBED/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/MAX32600MBED/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/MAX32600MBED/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
```

#### Target(s) affected by this defect ?

- MAX32600MBED
- MAX32620FTHR
- MAX32625MBED
- MAX32625PICO
- MAX32630FTHR

GCC and ARM Errors

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@ARMmbed/team-maximintegrated

talorion, 2020-08-26:
can this be closed?
works on my machine
mbed os: afcf91f3314d82ce8d555f4a891c9fd8db13910f
[MAX32600MBED.txt](https://github.com/ARMmbed/mbed-os/files/5132322/MAX32600MBED.txt)
[MAX32620FTHR.txt](https://github.com/ARMmbed/mbed-os/files/5132323/MAX32620FTHR.txt)
[MAX32625MBED.txt](https://github.com/ARMmbed/mbed-os/files/5132324/MAX32625MBED.txt)
[MAX32625PICO.txt](https://github.com/ARMmbed/mbed-os/files/5132325/MAX32625PICO.txt)
[MAX32630FTHR.txt](https://github.com/ARMmbed/mbed-os/files/5132326/MAX32630FTHR.txt)

MarceloSalazar, 2020-11-04:
Issue remains on Arm Compiler

ozersa, 2020-11-11:
I believe it can be closed?
Tested with mbed-os 6.4:  https://github.com/ARMmbed/mbed-os/commit/8ef0a435b2356f8159dea8e427b2935d177309f8
GCC version: 9 2019-q4-major
Boards: MAX32625MBED, MAX32630FTHR

It works.

MarceloSalazar, 2020-11-11:
GCC is known to work ok.
You may want to test AC6 (from Mbed Studio terminal) and comment if it compiles for you.

ozersa, 2020-11-11:
Thanks Marcelo for clarify, got it, it seems some update need to be done on ARM, IAR linker scripts for MAXIM boards.

PR #14465 (2021-04-15): Maxim: Add bare metal support to targets

### Summary of changes 

Fixes #13139 

Add heap load region to scatter file and update `targets.json` to
provide baremetal support to the following targets: SDT32620B,
SDT32625B, MAX32620FTHR, MAX32625MBED, MAX32625PICO, 
and MAX32630FTHR.

#### Impact of changes 
Maxim targets

#### Migration actions required 
None

### Documentation 

None

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / [... cut]
closed 2021-04-15 by 0xc0170