ARMmbed/mbed-os #13140: Few NXP targets don't compile with blinky-baremetal

MarceloSalazar, 2020-06-17:
### Description of defect

Few Silabs targets fail to compile blinky-baremetal in Mbed OS 6

```
Link: mbed-os-example-blinky-baremetal
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/LPC54114/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/LPC54114/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/LPC54114/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/LPC54114/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/LPC54114/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/LPC54114/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
```

#### Target(s) affected by this defect ?

- FF_LPC546XX
- LPC54114
- LPC546XX

GCC and ARM Errors

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@ARMmbed/team-nxp

talorion, 2020-08-26:
[FF_LPC546XX.txt](https://github.com/ARMmbed/mbed-os/files/5132288/FF_LPC546XX.txt)
[LPC546XX.txt](https://github.com/ARMmbed/mbed-os/files/5132290/LPC546XX.txt)
[LPC54114.txt](https://github.com/ARMmbed/mbed-os/files/5132291/LPC54114.txt)

can this be closed?
works on my machine
mbed os: afcf91f3314d82ce8d555f4a891c9fd8db13910f

MarceloSalazar, 2020-11-04:
Issue remains for FF_LPC546XX and Arm Compiler (works ok GCC)