ARMmbed/mbed-os #13146: NU_PFM_M2351_NPSA_S doesn't compile with normal blinky

MarceloSalazar, 2020-06-17:
### Description of defect

NU_PFM_M2351_NPSA_S fails to compile [blinky](github.com/ARMmbed/mbed-os-example-blinky) application with latest version of Mbed OS 6 (master)

 GCC and ARM error

```
Error: L6218E: Undefined symbol utest::v1::default_handlers (referred from BUILD/NU_PFM_M2351_NPSA_S/ARM/mbed-os/features/frameworks/utest/source/utest_harness.o).
Error: L6218E: Undefined symbol flash_set_target_config (referred from BUILD/NU_PFM_M2351_NPSA_S/ARM/mbed-os/hal/TARGET_FLASH_CMSIS_ALGO/flash_common_algo.o).
Error: L6218E: Undefined symbol CLK_DisableModuleClock_S (referred from BUILD/NU_PFM_M2351_NPSA_S/ARM/mbed-os/targets/TARGET_NUVOTON/TARGET_M2351/analogin_api.o).
Error: L6218E: Undefined symbol CLK_EnableModuleClock_S (referred from BUILD/NU_PFM_M2351_NPSA_S/ARM/mbed-os/targets/TARGET_NUVOTON/TARGET_M2351/analogin_api.o).
Error: L6218E: Undefined symbol CLK_SetModuleClock_S (referred from BUILD/NU_PFM_M2351_NPSA_S/ARM/mbed-os/targets/TARGET_NUVOTON/TARGET_M2351/analogin_api.o).
<...>
```

#### Target(s) affected by this defect ?

NU_PFM_M2351_NPSA_S 

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6.13 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m NU_PFM_M2351_NPSA_S 
```

@ARMmbed/team-nuvoton

ccli8, 2020-06-18:
Since TZ secure build support is removed from Mbed OS 6.0, remove NU_PFM_M2351_NPSA_S/NS from master (#13148). NU_PFM_M2351_NPSA_S/NS are kept on Mbed OS 5.15.

MarceloSalazar, 2020-06-18:
@ccli8 thanks for the quick fix!
I'm closing this issue as the PR is merged.

closed 2020-06-18 by MarceloSalazar