ARMmbed/mbed-os #13153: USB STM32F429 Discovery Board

narinaviation, 2020-06-18:
Hello, I'm trying to use the **STM32F429 Discovery Board** 
With **USB CDC/USB MSD** feature, Like when It works perfectly on my FRDM-K64F.
I have tried the example codes from https://os.mbed.com/docs/mbed-os/v6.0/apis/usbcdc.html
and https://os.mbed.com/docs/mbed-os/v6.0/apis/usbmsd.html
It then compiles without any error, and started to run.
BUT!!! **When I plug the Micro USB cable to the USB USER port on the other side of the board.
There is nothing happened. No "Found new hardware", No device connected sound. 
and the device does not appear in the "Device Manager" (The Windows can't detect it)**
Suddenly, I found that the **Green LED on PG13 was blinking without any code to do it.**
So, I try to do the debugging. And I see that **it goes into the mbed_die() function.**
Anyway, **When I use STM32CubeMX + MDK-ARM v5.27 to create the USB CDC Application.
The device is detected and worked correctly!!!** So I can confirm that it is not a hardware issues.
The target board is STM32F429-DISCO (Haven't tried it on STM32F429-DISC1 yet)
I've tested it on both the Mbed-OS 5 and Mbed-OS 6 (master)
both on Mbed Online Compiler and Mbed Studio.

0xc0170, 2020-06-19:
@narinaviation Please keep the issue template with the information

cc @ARMmbed/team-st-mcd

narinaviation, 2020-06-22:
> @narinaviation Please keep the issue template with the information
> 
> cc @ARMmbed/team-st-mcd

I'm very sorry, I've updated it now.

narinaviation, 2020-06-24:
Here, I come to supply more details about the error message found in serial monitor.

```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0: 0
R1: 0
R2: 80003CC
R3: 0
R4: 200006E0
R5: 0
R6: 200006BC
R7: 80
R8: 1
R9: 20001C18
R10: 0
R11: 0
R12: 0
SP   : 20001478
LR   : 8001FBD
PC   : 8003F7A
xPSR : 61000000
PSP  : 20001410
MSP  : 2002FFD0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 82
BFSR : 0
UFSR : 0
DFSR : 0
AFSR : 0
MMFAR: 8
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x8003F7A
Error Value: 0x20001CB0
Current Thread: main Id: 0x2000209C Entry: 0x800F18D StackSize: 0x1000 StackMem: 0x20000AF0 SP: 0x20001478 
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=DISCO_F429ZI
-- MbedOS Error Info --
```

jeromecoutant, 2020-06-24:
Current default DISCO_F429ZI configuration doesn't support USB_DEVICE.
Any help to support USB is welcome.
Starting point is to add "USBDEVICE" in "device_has_add" in targets.json file

narinaviation, 2020-06-24:
@jeromecoutant I've tried adding the "USBDEVICE" in the "device_has_add" section of the targets.json file
The result is still the same, then i try and into the mbed_app.json file, like this...
```
{
    "target_overrides" : {
        "*" : {
            "target.device_has_add" : ["USBDEVICE"]
        }
    }
}
```
Still not working, With the same error code 0x80FF013D.
Could you please let me know some working case configuration of the USBDEVICE on  STM32F429-DISCO? please.

0xc0170, 2020-06-25:
As mentioned by @jeromecoutant , the target does not support USB_DEVICE. Thus if you enable it, it might fail. More work is needed, to fix outstanding issues you are seeing. The starting point would be what triggers hardfault and fix the rootcause.

jeromecoutant, 2020-06-25:
> More work is needed, to fix outstanding issues you are seeing.

Yes...
I can share my dev branch as a starting point : https://github.com/jeromecoutant/mbed/tree/DEV_STM32_USB

0xc0170, 2020-07-02:
I'll close this issue , we can keep the discussion ongoing or move rather to a forum for this type of request.

closed 2020-07-02 by 0xc0170