ARMmbed/mbed-os #13157: Testing for changes when master and latest are out of sync

malavikasajikumar, 2020-06-18:
### Description of defect

Hello,

On Mbed CLI, a new project is created with the "latest" tagged mbed-os. If this is out of sync with the master branch (which is the case now), which version should the CI tests be run before submitting a pull request?

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-06-19:
`latest` points to the latest released version of Mbed OS. Master is moving target and marked once the release is read (tagged).

```
master ----------------------- | --------------- | ----------------- | ---------
                             6.1                6.1.1                6.2 = latest               
```

0xc0170, 2020-06-19:
The code picture should illustrate how master moves and it's being tagged.  There can be a situation when the latest produces the different result than master so worth testing as well.

malavikasajikumar, 2020-06-19:
Thanks, @0xc0170. Since there are modifications to the master branch than the last tagged as "latest", it is probably best to test against the master branch. However, wouldn't that cause error on the Jenkins CI tests run for every PR? Aren't those tests also run against "latest" tagged mbed-os?

0xc0170, 2020-06-22:
PRs targeting master are tested with merged master. PRs targeting 5.15 branch, are tested with merged mbed-os-5.15 master.

I hope that clarifies it. I'll close this issue. If you got any questions, just ask here.

closed 2020-06-22 by 0xc0170