ARMmbed/mbed-os #13162: I2C not working on NUCLEO-L4R5ZI

wdx04, 2020-06-21:
### Description of defect

On NUCLEO-L4R5ZI boards, a Mbed OS program can not communicate with I2C devices.
Tested devices are SHT20/SHT30 sensors and a SSD1306 OLED display with I2C interface.
All read/write operations to the connected I2C devices returns 1.
I have tested the same program on a few other NUCLEO boards, with the same I2C devices and wiring, all work without a problem.
The tested boards are:
NUCLEO-H743ZI2
NUCLEO-G071RB
NUCLEO-L432KC
NUCLEO-L476RG
NUCLEO-F767ZI
Also I wrote a similar program using the Arduino Framework, which works with all above boards including NUCLEO-L4R5ZI, so it's probably not a hardware issue.

#### Target(s) affected by this defect ?

NUCLEO-L4R5ZI

#### Toolchain(s) (name and version) displaying this defect ?

default ARMC6 toolchain with Mbed Studio

#### What version of Mbed-os are you using (tag or sha) ?

Mbed-OS 5.15 and 6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

MBed Studio 0.96 and 1.0

#### How is this defect reproduced ? 

Connect any I2C device to the default I2C pins D14/D15 and run a simple I2C scan program.
No I2C device can be found.

0xc0170, 2020-06-22:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-06-22:
I agree...

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/i2c_device.h#L55

SystemCoreClock  for STM32L4Rxx are 120MHz, which is not specified yet :-(

wdx04, 2020-06-23:
@jeromecoutant 
After applying your patch,  I2C on NUCLEO-L4R5ZI now works, thanks.

PR #13173 (2020-07-01): STM32L4: I2C init parameters for L4+ MCU
### Summary of changes 

SystemCoreClock for STM32L4Rxx are 120MHz, which was not specified in STM32L4 I2C init procedure.

#### Impact of changes 

Fix #13162 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking
closed 2020-07-01 by 0xc0170