ARMmbed/mbed-os #13163: ble.gap().getConnectionHandle() not defined

mkarn3yru96, 2020-06-21:
I am trying to update the connection parameters in my ble connection. But it's giving me an error that I am not able to solve...

The error:

**Error: Class "Gap"  has no member "getConnectionHandle" in "main.cpp", Line: 111, Col: 29**

#### Target(s) affected by this defect ?
NRF52 DK

#### How is this defect reproduced ? 

`    Gap::ConnectionParams_t *connParams;
    connParams->minConnectionInterval=250;
    connParams->maxConnectionInterval=400;
    connParams->slaveLatency=4;
    connParams->connectionSupervisionTimeout=550;
    connHandle = ble.gap().getConnectionHandle();////-------->>>
    ble_error_t resultCall= ble.updateConnectionParams(connHandle, connParams);`

0xc0170, 2020-06-22:
`getConnectionHandle` is not a member of Gap, is it?

Also the code snippet is not complete above, does not show where (and how) is ble defined.

Strictly speaking, this looks like an user error than technical bug in Mbed OS. Please provide more details to be certain.

mkarn3yru96, 2020-06-22:
> `getConnectionHandle` is not a member of Gap, is it?
**Indeed it is not.
How should I get the current connection handle? any help?**
> 
> Also the code snippet is not complete above, does not show where (and how) is ble defined.
> 
> Strictly speaking, this looks like an user error than technical bug in Mbed OS. Please provide more details to be certain.

**Here is the main.cpp : (how do I get the connection handle?) **

`#include "mbed.h"
#include "ble/BLE.h"
#include "ble/services/HeartRateService.h"
#include "ble/services/BatteryService.h"
#include "ble/services/DeviceInformationService.h"
#include "ble/services/HealthThermometerService.h"
#include "ECGService.h"

DigitalOut led1(p17);
DigitalOut led2(p20);

    const char ECG_UUID[] = "485f4145-52b9-4644-af1f-7a6b9322490f";
    const char ECG_SERVICE_UUID[] = "51311102-030e-485f-b122-f8f381aa84ed";

const static char     DEVICE_NAME[]        = "WOWpatch";
static const uint16_t uuid16_list[]        = {GattService::UUID_HEART_RATE_SERVICE,
                                              GattService::UUID_DEVICE_INFORMATION_SERVICE};
static volatile bool  triggerSensorPolling = false;
static volatile bool  triggerEcgPolling = false;

uint8_t hrmCounter = 100; // init HRM to 100bps
uint8_t batteryLevel =50; //init battery level to 50%
float temperature=28.10;
int16_t ecgValue=999;

HeartRateService         *hrService;
DeviceInformationService *deviceInfo;
BatteryService           *batteryService;
HealthThermometerService *thermometerService;
EcgService               *ecgService;

Gap::Handle_t connHandle;

void disconnectionCallback(const Gap::DisconnectionCallbackParams_t *params)
{
    BLE::Instance(BLE::DEFAULT_INSTANCE).gap().startAdvertising(); // restart advertising
}

void periodicCallback1(void) //ALL SENSORS EXCLUDING ECG
{
    led1 = !led1; /* Do blinky on LED1 while we're waiting for BLE events */

    /* Note that the periodicCallback() executes in interrupt context, so it is safer to do
     * heavy-weight sensor polling from the main thread. */
    triggerSensorPolling = true;
}

void periodicCallback2(void) // ECG
{
    led2 = !led2; /* Do blinky on LED2 while we're waiting for BLE events */

    /* Note that the periodicCallback() executes in interrupt context, so it is safer to do
     * heavy-weight sensor polling from the main thread. */
    triggerEcgPolling = true;
}

void bleInitComplete(BLE::InitializationCompleteCallbackContext *params)
{
    BLE &ble          = params->ble;
    ble_error_t error = params->error;
[... 75 lines cut ...]
                batteryLevel = 70;
            }
            batteryService->updateBatteryLevel(batteryLevel);
            
            temperature=temperature+0.11;
            if (temperature > 29.01) { //  
                temperature = 27.32;
            }
            thermometerService->updateTemperature(temperature);
            
        } 
        else if (triggerEcgPolling && ble.getGapState().connected) { //check periodick callback2 trigger   //////----------------------------------------->> if other callback is activated first, it never entrs here :(  ADD CONDITION
            triggerEcgPolling = false;

            ecgValue--;
            if (ecgValue == 800) { // 
                ecgValue = 999;
            }
            ecgService->updateEcgData(ecgValue);
        }
        else {
            ble.waitForEvent(); // low power wait for event
        }
    }
}`

pan-, 2020-06-22:
@mkarn3yru96 Ble controllers can handle several concurrent connections. It wouldn't be correct to ask for the current connection handle if many are active. 

The connection handle, is passed to the user in a [ConnectionCompleteEvent](https://github.com/ARMmbed/mbed-os/blob/b4389c7805d6a9547e760dafea3e1b1034d3d6ed/features/FEATURE_BLE/ble/gap/Events.h#L237). which is generated when the device is connected. 

Usage is demonstrated in the [ble-example](https://github.com/ARMmbed/mbed-os-example-ble)

0xc0170, 2020-06-22:
Thanks @pan- 

I'll close this. If you have any further questions, please use forum.mbed.com

closed 2020-06-22 by 0xc0170

mkarn3yru96, 2020-06-23:
"Usage is demonstrated in the ble-example "  in which of the examples?