ARMmbed/mbed-os #13167: Nightly: attestation example failing for NUCLEO_F429ZI

0xc0170, 2020-06-22:
### Description of defect

The previous run from the last week (559 job) all fine, but the recent one failed. See http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/561/example-test/315/FAIL/NUCLEO_F429ZI/ logs for details.

```
[1592611544.92][GLRM][INF] remote resources reset...
[1592611548.32][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1592611548.32][CONN][WRN] skipping __sync packet (specified with --sync=0)
[1592611548.32][CONN][RXD] Get attestation token:
[1592611548.32][CONN][RXD] 	assertion failed at ./main.cpp:91 - actual:-139expected:0
[1592611605.38][HTST][INF] test suite run finished after 150.95 sec...
[1592611605.38][HTST][ERR] Expected output [success\!] not received in log.
[1592611605.38][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1592611605.38][urllib3.connectionpool]Starting new HTTPS connection (1): mervi.mbedcloudtesting.com:443
[1592611605.56][urllib3.connectionpool]https://mervi.mbedcloudtesting.com:443 "PUT /resource/079602210531683D3A75FE96/disconnect HTTP/1.1" 200 62
[1592611608.93][HTST][INF] CONN exited with code: 0
[1592611608.93][HTST][INF] No events in queue
[1592611608.93][HTST][INF] stopped consuming events
[1592611608.93][HTST][INF] host test result(): None
[1592611608.93][HTST][WRN] missing __exit event from DUT
[1592611608.93][HTST][WRN] missing __exit_event_queue event from host test
[1592611608.93][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1592611608.93][HTST][INF] calling blocking teardown()
[1592611608.93][HTST][INF] teardown() finished
[1592611608.93][HTST][INF] {{result;timeout}}
```

I don't see a change in the example itself (also passes for other boards) so this means something changed in Mbed OS causing this example test to timeout.

cc @ARMmbed/team-st-mcd Please review

#### Target(s) affected by this defect ?

NUCLEO_F429ZI

#### Toolchain(s) (name and version) displaying this defect ?

Both
#### What version of Mbed-os are you using (tag or sha) ?

3d031cebe4c3b03eb7871ee62f9b96ee8faa1755

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Any

#### How is this defect reproduced ? 

use example test script

jeromecoutant, 2020-06-22:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L1967

I don't really know impact of this...
Came with #12955

0xc0170, 2020-06-22:
Thanks @jeromecoutant , I wanted to look at that PSA PR if that relates.

@Patater Please review - how can we fix the example failing?

jamesbeyond, 2020-06-25:
I think this got fixed. if you look at latest: 
http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/566/example-test/

0xc0170, 2020-06-25:
I'll close then. Not certain why it failed at the first place (might be target timeout).

closed 2020-06-25 by 0xc0170