ARMmbed/mbed-os #13170: ESP8266 WiFi example not working in Mbed OS 6.0

tatyanavolkova, 2020-06-22:
Hello all,

Found out that in Mbed 6.0. I keep getting the same error in standard mbed-os-example-wifi sample program. However, in Mbed 5.15.3 all works, with the same config, the ESP8266 finds the AP list. I have NonOS AT 1.7.4 firmware on ESP.

Log:
`/WiFi example
Mbed OS version 6.0.0

Scan:
ESP8266: ERROR: AT Firmware v-1 incompatible with this driver.Update at least to v1 - https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update

++ MbedOS Error Info ++
Error Status: 0x80080109 Code: 265 Module: 8
Error Message: Too old AT firmware
Location: 0x800608B
Error Value: 0x0
Current Thread: main Id: 0x200028D0 Entry: 0x800ED01 StackSize: 0x1000 StackMem: 0x20001670 SP: 0x200024FC 
For more info, visit: https://mbed.com/s/error?error=0x80080109&tgt=NUCLEO_L073RZ
-- MbedOS Error Info --`

0xc0170, 2020-06-23:
If you run the example from https://os.mbed.com/teams/ESP8266/wiki/Firmware-Update , section version firmware check - what is being printed - what version?

tatyanavolkova, 2020-06-23:
I have the following firmware version:

AT+GMR
AT version:1.7.4.0(May 11 2020 19:13:04)
SDK version:3.0.4(9532ceb)
compile time:May 27 2020 10:12:22
Bin version(Wroom 02):1.7.4
OK

Unfortunately the error message I mentioned above is not clear to me. It says that v2 firmware is required, as I understand. However, all tutorials and all documentations say that the example works with 1.7.0-1.7.4. firmware. There is no 2.0 firmware (NonOS AT) on espressif website, the only one I see is IDF frimware which is 2.0 and which is uncompatible with some ESP breakout boards because it uses different pins for communication and flashing

0xc0170, 2020-06-24:
It prints:

```
ESP8266: ERROR: AT Firmware v-1 incompatible with this driver.Update at least to v1 - https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update
```

This comes from:

```
bool ESP8266Interface::_get_firmware_ok()
{
    ESP8266::fw_at_version at_v = _esp.at_version();
    if (at_v.major < ESP8266_AT_VERSION_MAJOR) {
        debug("ESP8266: ERROR: AT Firmware v%d incompatible with this driver.", at_v.major);
        debug("Update at least to v%d - https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update\n", ESP8266_AT_VERSION_MAJOR);
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_DRIVER, MBED_ERROR_UNSUPPORTED), "Too old AT firmware");
    }
    ESP8266::fw_sdk_version sdk_v = _esp.sdk_version();
    if (sdk_v.major < ESP8266_SDK_VERSION_MAJOR) {
        debug("ESP8266: ERROR: Firmware v%d incompatible with this driver.", sdk_v.major);
        debug("Update at least to v%d - https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update\n", ESP8266_SDK_VERSION_MAJOR);
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_DRIVER, MBED_ERROR_UNSUPPORTED), "Too old SDK firmware");
    }

    return true;
}
```

Therefore `at_v.major` is `-1` isn't it ? Please check if that is the case . what version is returned from `_esp.at_version()`

tatyanavolkova, 2020-07-12:
> Therefore `at_v.major` is `-1` isn't it ? Please check if that is the case . what version is returned from `_esp.at_version()`

Well when I rty to do the following (to print values of at_v structure, which the _esp.at_version() returns):
debug("Current firmware is version v%d.%d.%d", at_v.major, at_v.minor, at_v.patch);
I see the following output:
-1.-1.-1

So there seems to be an error but I can't figure out how to fix it, especially when in MBed 5.0 all worked like charm

0xc0170, 2020-07-13:
@ARMmbed/mbed-os-ipcore

michalpasztamobica, 2020-07-13:
Hi @tatyanavolkova , I haven't seen the version set to -1 ever before, but what you can try is update the firmware on you esp, just in case. Last time I did this I used [this guide](https://os.mbed.com/users/sarahmarshy/notebook/esp8266-v2-firmware-update-/). Let us know if this helps?

adbridge, 2020-11-17:
@tatyanavolkova did you manage to update your firmware as per Michael's comment ?