ARMmbed/mbed-os #13176: NRF5x: Uninitalized variable in gpio_api leads to wrong configured GPIO.

supertoast, 2020-06-23:
### Description of defect

The function `gpio_apply_config` in gpio_api of NRF5x using a  variable with uninitalized memory witch leads to wrong configured GPIO. The member `skip_gpio_setup` is not initalized, so in our case the configuration of the pull up resistor was sometimes not applied.
[Source]https://github.com/ARMmbed/mbed-os/blob/e4b81f67f939a0c0b11c147ce74aa367271e1279/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/gpio_api.c#L127

#### Target(s) affected by this defect ?
We use a custom board with NRF52832. I think all targets with this MCU are affected.

#### Toolchain(s) (name and version) displaying this defect ?
GCC for Arm (gcc-arm-none-eabi-9-2019-q4-major)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
The error occurs randomly because the memory is uninitialized.

0xc0170, 2020-06-24:
@supertoast Thanks for the report. Can you send a new pull request fixing the problem?

supertoast, 2020-06-24:
Looks like the bug is already fixed.
https://github.com/ARMmbed/mbed-os/commit/e9dd6020b5d8f5673e4ed41a84990cc95de5ff2f

closed 2020-06-24 by supertoast

0xc0170, 2020-06-24:
@supertoast that is although on master branch. It might be useful to propagate to 5.15 branch as well? if yes, can you please backport it (send a new pull request) ?

supertoast, 2020-06-24:
Done. #13184