ARMmbed/mbed-os #13179: tests-mbed_hal-rtc tests in mbed os 6 fail for targets due to wrong assertion statement in code

SAGAR11111995, 2020-06-24:
### Description of defect

I tried testing my board for RTC tests. tests-mbed_hal-rtc test was failing for rtc_range_test. In rtc_range_test test case, assertion statement used is as below:

        TEST_ASSERT_DURATION_WITHIN(WAIT_TOLERANCE, WAIT_TIME, stop - start);

This macro function is defined as follows:
#define TEST_ASSERT_DURATION_WITHIN(delta, expected, actual) \
    do { \
        using ct = std::common_type_t<decltype(delta), decltype(expected), decltype(actual)>; \
        TEST_ASSERT_INT_WITHIN(ct(delta).count(), ct(expected).count(), ct(actual).count()); \
    } while (0)

The test case contains 4 time values which it is writing and reading from device's RTC module
These 4 values are: 
static const RealTimeClock::time_point starts[] = {
        RealTimeClock::time_point{0x00000000s},
        RealTimeClock::time_point{0xEFFFFFFFs},
        RealTimeClock::time_point{0x00001000s},
        RealTimeClock::time_point{0x00010000s},
    };
The Second value (0xEFFFFFFFs) is a 32 bit long unsigned integer and for doing comparison operations with it, Assertion statement for unsigned int 32 value type(TEST_ASSERT_UINT32_WITHIN) should be used as was used till mbed os 5 latest version. But in mbed os 6, instead of this statement, assertion statement for signed int type(TEST_ASSERT_INT_WITHIN) is used. Hence when comparing value like (0xEFFFFFFF) it is assuming it as 31 bit integer value with MSB as sign bit which should not happen. Hence assertion is failing and test is failing as well. When this(TEST_ASSERT_INT_WITHIN) statement was replaced with this(TEST_ASSERT_UINT32_WITHIN)  statement, test case passed.

Can you please help me with this issue?

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed cli 1.8.3

#### How is this defect reproduced ? 
Everytime tests are run

SAGAR11111995, 2020-06-24:
What target(s) are you using?
-> Irrespective of any board

What version of Mbed OS are you using (tag or sha)?
mbed-os-6.0.0( 165be79)

toyowata, 2020-06-25:
Reopen this as I am getting same issue with other targets.

0xc0170, 2020-06-25:
What was your previous version that was failing?

0xc0170, 2020-06-25:
@toyowata What version did you reproduce this with ? 6.x had some issues , previously reported by users. Not recommended to use and update is required.

toyowata, 2020-06-25:
I am using v9.3.1 GCC toolchain and the tests-mbed_hal-rtc failed with Renesas GR-MANGO target (Cortex-A9).

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The Arm compile 6.13 build works fine (tests-mbed_hal-rtc passed).  It may be Cortex-A9 code-gen related issue or something.  I think this issue #13179 can be closed and I will raise another issue when I get more information.

closed 2020-06-25 by SAGAR11111995