ARMmbed/mbed-os #13181: Calling memalign throws a precise data access error

janjongboom, 2020-06-24:
### Description of defect

We need to align some memory, and it looks like `posix_memalign` is not available in Mbed OS 5.15 with GCC ARM 9. Using `memalign` crashes the DISCO-L475VG-IOT01A development board directly. Here's an example:

```
#include <malloc.h>

int main() {
    void *ptr = memalign(16, 1024);
    printf("memalign post %p\n", ptr);
}
```

This crashes with a hard fault:

```
Crash Info:
	Crash location = _memalign_r [0x0801B2F8] (based on PC value)
	Caller location = __rtos_malloc_lock [0x0800F227] (based on LR value)
	Stack Pointer at the time of crash = [20002728]
	Target and Fault Info:
		Processor Arch: ARM-V7M or above
		Processor Variant: C24
		Forced exception, a fault with configurable priority has been escalated to HardFault
		A precise data access error has occurred. Faulting address: EEADD3A0
```

#### Target(s) affected by this defect ?

Tested on the ST IoT Discovery Kit (DISCO-L475VG-IOT01A target name). Compiled in debug mode.

#### Toolchain(s) (name and version) displaying this defect ?

gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 

#### What version of Mbed-os are you using (tag or sha) ?

64853b354fa188bfe8dbd51e78771213c7ed37f7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

See above.

0xc0170, 2020-06-24:
@ARMmbed/team-st-mcd would you know why memalign would fail?

0xc0170, 2020-06-30:
Have you made any progress with this failure ? What is the value of `BFAR` ? Is it `EEADD3A0` as reported? How come something tries to write to this address and produces this invalid memory access?

The heap address is completely different than `EEADD3A0` should in ranges 0x1.. or 0x2... based on SRAM1/SRAM2. I can see heap is for gcc in SRAM1

janjongboom, 2020-07-02:
@0xc0170 I have not. We've switched to a custom way of allocating aligned memory on the heap now, as we needed to get a release out that was depending on this. I assume @armmbed/team-st-mcd would be able to triage this much quicker.

JojoS62, 2022-11-13:
I think I ran into the same problem. aligned_alloc / memalign is now available in gcc >10.2 and it works. But it crashes with hardfault when it is used with the heap statistics.  memalign requests a larger block but manipulates the address, this is not compatible with the tracing and manipulating in the malloc_wrapper in platform/mbed_alloc_wrappers.cpp.
@0xc0170 reopen this issue or do you want a new one?

0xc0170, 2022-11-14:
Please open a new one with up to date information.