ARMmbed/mbed-os #13187: UHURU_RAVEN doesn't compile greentea tests for baremetal

MarceloSalazar, 2020-06-24:
### Description of defect

UHURU_RAVEN targets fails to compile greentea tests for baremetal in Mbed OS 6

```
Build failures:
  * UHURU_RAVEN::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-GENERAL_BLOCK_DEVICE
        Building project general_block_device (UHURU_RAVEN, GCC_ARM)
        Scan: GCC_ARM
        Scan: general_block_device
        Compile [100.0%]: main.cpp
        [Fatal Error] main.cpp@32,10: SPIFBlockDevice.h: No such file or directory
        
  * UHURU_RAVEN::GCC_ARM::FEATURES-STORAGE-TESTS-FILESYSTEM-GENERAL_FILESYSTEM
        Building project general_filesystem (UHURU_RAVEN, GCC_ARM)
        Scan: GCC_ARM
        Scan: general_filesystem
        Compile [100.0%]: main.cpp
        [Fatal Error] main.cpp@26,10: SPIFBlockDevice.h: No such file or directory
```

#### Target(s) affected by this defect ?

- UHURU_RAVEN

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
git clone https://github.com/ARMmbed/mbed-os
mbed deploy
mbed test -t GCC_ARM -m UHURU_RAVEN --compile --app-config mbed-os/TESTS/configs/baremetal.json -c
```

@jamesbeyond @toyowata @ARMmbed/team-st-mcd @ARMmbed/team-uhuru

toyowata, 2020-06-25:
@MarceloSalazar 
I could reproduce the compile error with `UHURU_RAVEN` target.
Other targets (`K82F`, `MTS_DRAGONFLY_F413RH` and `EP_ATRAS`) have same issue which has `COMPONENT_SPIF` definition.

I have added the `spif` value in the baremetal.json here, but I still got compile error.

https://github.com/ARMmbed/mbed-os/blob/a6207cadad0acd1876f436dc6baeddf46c42af06/TESTS/configs/baremetal.json#L2-L38

toyowata, 2020-06-25:
Ah, `spif` is not proper name of mbed_lib.json in the SPIF component.

https://github.com/ARMmbed/mbed-os/blob/a6207cadad0acd1876f436dc6baeddf46c42af06/components/storage/blockdevice/COMPONENT_SPIF/mbed_lib.json#L2

Adding the `"spif-driver"` in `baremetal.json` fixes the problem.

closed 2020-07-01 by 0xc0170