ARMmbed/mbed-os #13189: QSPIFBlockDevice fails to compile for STM32 NUCLEO_F767ZI

nat45928, 2020-06-25:
### Description of defect

I am attempting to add NOR flash memory to my project but the provided `QSPIFBlockDevice` component will not compile for the STM32F767ZI Nucleo target. I get the following error when compiling:

```
...
Building project test-gantry (NUCLEO_F767ZI, GCC_ARM)
Scan: project
Scan: mbed-os
Compile [ 71.0%]: PortInOut.cpp
Compile [ 71.2%]: PortIn.cpp
Compile [ 71.3%]: I2CSlave.cpp
Compile [ 71.4%]: PwmOut.cpp
Compile [ 71.5%]: PortOut.cpp
Compile [ 71.6%]: RawSerial.cpp
[Warning] RawSerial.cpp@52,21: 'int mbed::RawSerial::putc(int)' is deprecated: The class has been deprecated and will be removed in the future. [-Wdeprecated-declarations]
[Warning] RawSerial.cpp@66,40: 'int mbed::RawSerial::vprintf(const char*, va_list)' is deprecated: The class has been deprecated and will be removed in the future. [-Wdeprecated-declarations]
[Warning] RawSerial.cpp@81,18: 'int mbed::RawSerial::puts(const char*)' is deprecated: The class has been deprecated and will be removed in the future. [-Wdeprecated-declarations]
[Warning] RawSerial.cpp@85,18: 'int mbed::RawSerial::puts(const char*)' is deprecated: The class has been deprecated and will be removed in the future. [-Wdeprecated-declarations]
Compile [ 71.7%]: InterruptIn.cpp
Compile [ 71.8%]: QSPIFBlockDevice.cpp
[Error] mbed_config.h@41,79: 'QSPI_FLASH1_IO0' was not declared in this scope; did you mean 'QSPI_FLASH_ID_2'?
[Error] mbed_config.h@42,79: 'QSPI_FLASH1_IO1' was not declared in this scope; did you mean 'QSPI_FLASH_ID_1'?
[Error] mbed_config.h@43,79: 'QSPI_FLASH1_IO2' was not declared in this scope; did you mean 'QSPI_FLASH_ID_2'?
[Error] mbed_config.h@44,79: 'QSPI_FLASH1_IO3' was not declared in this scope; did you mean 'QSPI_FLASH_ID_2'?
[Error] mbed_config.h@45,79: 'QSPI_FLASH1_SCK' was not declared in this scope; did you mean 'QSPI_FLAG_TC'?
[Error] mbed_config.h@40,79: 'QSPI_FLASH1_CSN' was not declared in this scope; did you mean 'QSPI_FLASH_ID_2'?
...
```

It appears the default QSPI pin names

* `QSPI_FLASH1_IO0`
* `QSPI_FLASH1_IO1`
* `QSPI_FLASH1_IO2`
* `QSPI_FLASH1_IO3`
* `QSPI_FLASH1_SCK`
* `QSPI_FLASH1_CSN`

do not exist for the F7 as expected.

The names are needed by the [`QSPIFBlockDevice` constructor](https://github.com/ARMmbed/mbed-os/blob/master/components/storage/blockdevice/COMPONENT_QSPIF/QSPIFBlockDevice.h#L128) and appear to only be added to a few STM32 targets (like the STM32L4 [here](https://github.com/ARMmbed/mbed-os/commit/8cd00b346879388e83a32826c73629fbad14b2d4#diff-7eddf79cd973767758044ca9e8c8ae7bR355).

My `mbed_app.json` enables QSPI by adding:

```json
{
  "target_overrides": {
    "NUCLEO_F767ZI": {
      "target.device_has_add": ["QSPI"],
      "target.components_add" : ["QSPIF"]
    }
  }
}
```

Is adding the following snippet to https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI/TARGET_NUCLEO_F767ZI/PinNames.h#L364 an acceptable fix?

```
    /**** QSPI FLASH pins ****/
    QSPI_FLASH1_IO0 = PD_11,
    QSPI_FLASH1_IO1 = PD_12,
    QSPI_FLASH1_IO2 = PE_2,
    QSPI_FLASH1_IO3 = PD_13,
    QSPI_FLASH1_SCK = PB_2,
    QSPI_FLASH1_CSN = PB_6,
```

#### Target(s) affected by this defect ?

NUCLEO_F767ZI

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM, 9-2019-q4-major 9.2.1 20191025

#### What version of Mbed-os are you using (tag or sha) ?

#d5f034b637957f9123e2a83257439eb3e1d80ca7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

Attempt to compile QSPI and the `QSPIFBlockDevice` component for the `NUCLEO_F767ZI` target.

nat45928, 2020-06-25:
It just dawned on me that the `QSPI_FLASH1_*` names are describing a QSPI based flash device, not the QSPI peripheral. That flash device is not present on a Nucleo, which is why the names are not present.

What is the right way to override the `PinNames.h` in a custom target to add this functionality?

jeromecoutant, 2020-06-25:
Hi
If you add QSPI to device_has or component, you also have to specify the pins in your mbed_app.json:

```
{
  "target_overrides": {
    "NUCLEO_F767ZI": {
      "target.device_has_add": ["QSPI"],
      "target.components_add" : ["QSPIF"],
      "drivers.qspi_io0": "PX_x",
...
      "drivers.qspi_csn": "PX_x"
    }
  }
}
```

0xc0170, 2020-06-25:
> If you add QSPI to device_has or component, you also have to specify the pins in your mbed_app.json:

Does that answer it @nat45928  ?

jeromecoutant, 2020-06-25:
@MarceloSalazar @0xc0170 
This should be documented in doc ?
https://os.mbed.com/docs/mbed-os/v6.0/apis/spi-apis.html

nat45928, 2020-06-25:
That did the trick! Thanks @jeromecoutant!

It definitely wasn't clear that those definitions were required (and the right thing to set) from the docs on the `QSPIFBlockDevice` and the `mbed_lib.json` configuration.

0xc0170, 2020-06-26:
@nat45928 would you be able to improve the documentation via new pull request? This is the repository where SPI API is described

jeromecoutant, 2020-07-02:
Issue raised in doc part: https://github.com/ARMmbed/mbed-os-5-docs/issues/1347

@0xc0170 could you close this one ?
Thx

closed 2020-07-02 by 0xc0170