ARMmbed/mbed-os #13195: mbed_drivers-buffered_serial Test fails for SDP-K1

malavikasajikumar, 2020-06-25:
### Description of defect

mbed-os-tests-mbed_drivers-buffered_serial test fails on SDP-K1 for the **master** branch mbed-os. Recent modifications were made to the STM_32 Family definition. Before these edits, this test was not run for SDP-K1.  
The error says "SYNC FAILED". 

#### Target(s) affected by this defect ?
SDP-K1 uses STM32F469NI

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli --version
1.10.4

#### How is this defect reproduced ? 
1.	mbed new mbed_test_master
2.	Git switch branch to master on mbed-os
3.	mbed test -t GCC_ARM -m SDP_K1 -n mbed-os-tests-mbed_drivers-buffered_serial -v

[SDP_K1_GCC_ARM_master_serial_test.log](https://github.com/ARMmbed/mbed-os/files/4833212/SDP_K1_GCC_ARM_master_serial_test.log)

0xc0170, 2020-06-26:
> mbed-os-tests-mbed_drivers-buffered_serial test fails on SDP-K1 for the master branch mbed-os. Recent modifications were made to the STM_32 Family definition. Before these edits, this test was not run for SDP-K1.

Means the bug has been there for longer, correct?

cc @ARMmbed/team-st-mcd

0xc0170, 2020-06-30:
> mbed-os-tests-mbed_drivers-buffered_serial test fails on SDP-K1 for the master branch mbed-os. Recent modifications were made to the STM_32 Family definition. Before these edits, this test was not run for SDP-K1.
The error says "SYNC FAILED".

One more question, what is the last revision (SHA) that worked?

malavikasajikumar, 2020-07-09:
> One more question, what is the last revision (SHA) that worked?

SHA-1: b6370b4c37f3d4665ed1cdcb1afea85396bba1b3

malavikasajikumar, 2020-07-09:
> > mbed-os-tests-mbed_drivers-buffered_serial test fails on SDP-K1 for the master branch mbed-os. Recent modifications were made to the STM_32 Family definition. Before these edits, this test was not run for SDP-K1.
> 
> Means the bug has been there for longer, correct?
> 
> cc @ARMmbed/team-st-mcd

@0xc0170 Not sure if this bug was there for longer, but with the "latest" tagged mbed-os, **mbed-os-tests-mbed_drivers-buffered_serial** test gets skipped for SDP-K1. For the master branch, it biulds and tried to run the test, and reports a SYNC_FAILED.

Also, the FAMILY_STM32 definition was modified on targets.json recently, and corresponding changes were made to the SDP-K1 portion. These changes were not from ADI. How can we make sure the ADI team is notified from the Mbed side of any such changes in the future. We are working with @jeromecoutant to make sure any changes are tested on SDP-K1 as well. However, what is Mbed's policy of informing Board vendors when board support package of its Target vendor gets modified? @andypowers

malavikasajikumar, 2020-07-15:
> @malavikasajikumar it has been 5 days since the last reminder. Could you please update the issue header as previously requested?

Hello, this has been fixed. Please confirm.

malavikasajikumar, 2020-07-19:
@0xc0170 @jamesbeyond  We have a few modifications to make to the SDP-K1 board support package that are stable on the "latest" tag mbed-os. How should we submit the changes? Does it still make sense to submit the changes to master branch where one of the tests fail?

malavikasajikumar, 2020-07-21:
https://github.com/ARMmbed/mbed-os/pull/13320

closed 2020-07-21 by malavikasajikumar

0xc0170, 2020-07-21:
👍  if any further question, just ask in the 13157 issue