ARMmbed/mbed-os #13197: Non-blocking recv() isn't really non-blocking with lwip stack

multiplemonomials, 2020-06-26:
### Description of defect

In my project I need to implement a system with a high update rate (2kHz) that communicates over Ethernet.  So far I have sending working fine, but when I attempted receiving I ran into trouble:  If there is currently no data in the receive buffer, it seems like UDPSocket::recvfrom() will block for at least 500 us!  

I investigated the problem some more and I think I see what's going on.  It looks like LWIPStack always creates LWIP sockets in blocking mode with a timeout of 1ms, regardless of what the Socket's blocking mode is set to.  So, recv() will always wait about a millisecond for data to arrive before returning.

It seems like the best fix for this would be for LWIPStack to correctly set non-blocking sockets to be non-blocking in LWIP.  Another, related issue is that blocking sockets still only use a timeout of 1, so they turn into a resource-intensive spinlock.  It looks like this could be avoided pretty easily by proxying the actual timeout value to lwip, so it will then mutex wait for the correct amount of time before returning.

#### Patch:
There are probably a lot of changes needed to properly handle non-blocking sockets, so I came up with a simple (but dirty) patch:
In sys_arch_mbox_fetch() in lwip_sys_arch.c, I changed:
```c
    uint32_t flags = osEventFlagsWait(mbox->id, SYS_MBOX_FETCH_EVENT,
            osFlagsWaitAny | osFlagsNoClear, (timeout ? timeout : osWaitForever));
```
to
```c
    uint32_t osTimeout;
    if(timeout == 0)
    {
        // infinite wait
        osTimeout = osWaitForever;
    }
    else if(timeout == 1)
    {
        // PATCH: treat 1ms timeout as 0
        osTimeout = 0;
    }
    else
    {
        osTimeout = timeout;
    }

    uint32_t flags = osEventFlagsWait(mbox->id, SYS_MBOX_FETCH_EVENT,
            osFlagsWaitAny | osFlagsNoClear, osTimeout);
```

Since LWIPInterface uses a timeout of 1 everywhere, this will change it to a timeout of 0, so the mutex wait will return immediately if there is no message.  We need to do this here because otherwise it changes the 0 value to osWaitForever.

In my testing this brings the recv() delay down to where it should be, about 10 us.  It will prevent blocking sockets from working properly though so keep that in mind.  I also haven't tested it with TCP.

#### Target(s) affected by this defect ?

I was testing on a NUCLEO_F429ZI, but I think that this affects all targets.

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM 2019 q4

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

I'm using my custom build system, https://github.com/USCRPL/mbed-cmake

This integrates with the MBed python scripts, and should have the same build behavior as mbed-cli.

#### How is this defect reproduced ? 

1. Create a UDP socket in non-blocking mode.  Don't send any data to it.
2. Call `socket.recvfrom()`.

**Expected behavior:** With no data in the socket I expect recvfrom() to take no more than double digit numbers of microseconds to run.

**Actual behavior:** recvfrom() takes about 480-500ms to execute.

0xc0170, 2020-06-29:
cc @ARMmbed/mbed-os-ipcore

kjbracey-arm, 2020-06-29:
Thanks for identifying this.

I was unaware of this - I was convinced we were using non-blocking calls to lwIP.

Indeed, TCP sockets are set to non-blocking.

This is related to  https://github.com/ARMmbed/mbed-os/issues/13056, where there was confusion because I thought that sockets were created non-blocking, and that the lines in `connect` were temporarily setting to blocking and putting it back.

When I realised that wasn't the case, and `connect` was setting it non-blocking for the first time, alarm bells should have gone off.

If `connect` is setting TCP sockets non-blocking for the first time, then who is setting UDP sockets non-blocking? No-one, apparently. Eep!

> Another, related issue is that blocking sockets still only use a timeout of 1, so they turn into a resource-intensive spinlock if recv() is called with a longer timeout. 

Not quite following this comment - all `LWIPStack` calls are supposed to always be non-blocking, regardless of the nsapi-level blocking/timeout setting, and `InternetDatagramSocket::recvfrom` will wait for a call to the `socket_attach` callback before retrying, if it wants to block.

I would hope that just replacing that `netconn_set_recvtimeout(s->conn, 1)` with `netconn_set_nonblocking(s->conn, true)` should work. TCP sockets are operating in that mode already.

multiplemonomials, 2020-06-29:
Alright I tried that change and everything seems to be working.

> Another, related issue is that blocking sockets still only use a timeout of 1, so they turn into a resource-intensive spinlock if recv() is called with a longer timeout.

Scratch that, it was just my misunderstanding how that part of the socket code works plus an issue with my test code causing bad performance.  Everything seems to be fine with blocking sockets.

0xc0170, 2020-06-30:
@multiplemonomials Thanks for detailed report and @kjbracey-arm for the instant fix

PR #13205 (2020-07-01): LWIPStack: set sockets non-blocking

### Summary of changes 

When sockets were opened, they were not set non-blocking, but only had their timeout set to 1ms.

TCP sockets were set to non-blocking on connect or accept, but UDP sockets were left blocking with 1ms timeout.

This led to unnecessary delays when checking for data from an empty UDP socket.

Change sockets to be non-blocking on open instead of setting the timeout. As TCP sockets were already non-blocking, the necessary event handling is already in place, and no other changes are required.

#### Impact of changes 

Fixes #13197

####  [... cut]
closed 2020-07-01 by 0xc0170